serialport/node-serialport #1997: Rebuild failed on windows

AlexandreMouyen, 2019-12-08:
Hello,

I spent all my day to try to rebuild my app because I need to use the serialport package (NODE_MODULE_VERSION are different) but I cannot find a solution.

I am working on Windows 10, electron: 7.1.3, electron-rebuild: 1.8.8.

```
An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@6.0.1
gyp info using node@12.13.1 | win32 | x64
gyp info find Python using Python version 2.7.17 found at "C:\Python27\python.exe"
gyp info find VS using VS2017 (15.9.28307.905) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Python27\python.exe
gyp info spawn args [
gyp info spawn args   'D:\\Documents\\space-controller\\node_modules\\node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\Documents\\space-controller\\node_modules\\@serialport\\bindings\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\Documents\\space-controller\\node_modules\\node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\gegem\\.electron-gyp\\7.1.3\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\gegem\\.electron-gyp\\7.1.3',
gyp info spawn args   '-Dnode_gyp_dir=D:\\Documents\\space-controller\\node_modules\\node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\gegem\\\\.electron-gyp\\\\7.1.3\\\\<(target_arch)\\\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=D:\\Documents\\space-controller\\node_modules\\@serialport\\bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'D:\\Documents\\space-controller\\node_modules\\@serialport\\bindings\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
Les projets sont g�n�r�s individuellement dans cette solution. Pour activer la g�n�ration en parall�le, ajoutez le commutateur "/m".
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(8513): warning C4996: 'v8::MicrotasksCompletedCallback': a �t� d�clar� d�conseill� (compilation du fichier source ..\src\serialport_win.cpp) [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
  c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(6918): note: voir la d�claration de 'v8::MicrotasksCompletedCallback' (compilation du fichier source ..\src\serialport_win.cpp)
c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(8522): warning C4996: 'v8::MicrotasksCompletedCallback': a �t� d�clar� d�conseill� (compilation du fichier source ..\src\serialport_win.cpp) [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
  c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(6918): note: voir la d�claration de 'v8::MicrotasksCompletedCallback' (compilation du fichier source ..\src\serialport_win.cpp)
c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(8513): warning C4996: 'v8::MicrotasksCompletedCallback': a �t� d�clar� d�conseill� (compilation du fichier source ..\src\serialport.cpp) [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
  c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(6918): note: voir la d�claration de 'v8::MicrotasksCompletedCallback' (compilation du fichier source ..\src\serialport.cpp)
c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(8522): warning C4996: 'v8::MicrotasksCompletedCallback': a �t� d�clar� d�conseill� (compilation du fichier source ..\src\serialport.cpp) [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
  c:\users\gegem\.electron-gyp\7.1.3\include\node\v8.h(6918): note: voir la d�claration de 'v8::MicrotasksCompletedCallback' (compilation du fichier source ..\src\serialport.cpp)
d:\documents\space-controller\node_modules\@serialport\bindings\src\serialport.cpp(329): error C2661: 'v8::Object::Set'�: aucune fonction surcharg�e ne n�cessite 2 arguments [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
d:\documents\space-controller\node_modules\@serialport\bindings\src\serialport.cpp(330): error C2661: 'v8::Object::Set'�: aucune fonction surcharg�e ne n�cessite 2 arguments [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
d:\documents\space-controller\node_modules\@serialport\bindings\src\serialport.cpp(331): error C2661: 'v8::Object::Set'�: aucune fonction surcharg�e ne n�cessite 2 arguments [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
d:\documents\space-controller\node_modules\@serialport\bindings\src\serialport.cpp(378): error C2661: 'v8::Object::Set'�: aucune fonction surcharg�e ne n�cessite 2 arguments [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (D:\Documents\space-controller\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\Documents\\space-controller\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=7.1.3" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
[... 61 lines cut ...]
d:\documents\space-controller\node_modules\@serialport\bindings\src\serialport.cpp(378): error C2661: 'v8::Object::Set'�: aucune fonction surcharg�e ne n�cessite 2 arguments [D:\Documents\space-controller\node_modules\@serialport\bindings\build\bindings.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (D:\Documents\space-controller\node_modules\node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.18362
gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "D:\\Documents\\space-controller\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild" "--target=7.1.3" "--arch=x64" "--dist-url=https://electronjs.org/headers" "--build-from-source"
gyp ERR! cwd D:\Documents\space-controller\node_modules\@serialport\bindings
gyp ERR! node -v v12.13.1
gyp ERR! node-gyp -v v6.0.1
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (D:\Documents\space-controller\node_modules\spawn-rx\lib\src\index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.error (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:156:26)
    at Subscriber._error (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:92:26)
    at Subscriber.error (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:72:18)
    at SafeSubscriber._next (D:\Documents\space-controller\node_modules\spawn-rx\lib\src\index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.next (D:\Documents\space-controller\node_modules\rxjs\internal\Subscriber.js:143:22)
```

frozenex, 2019-12-16:
Having the same issue +1

webtop, 2019-12-17:
I have this same issue on a number of projects that use serialport.

webtop, 2019-12-17:
@AlexandreMouyen @frozenex 
If you have the Windows version of NodeJs installed, uninstall it.
If you do not have nvm-windows installed, install it.

`nvm install 9`

Once the download has finished

`nvm use 9.11.2 (or whatever version it spits out)`

In your application folder try to rebuild.

Seems that serialport isn't compiling with versions of node > 9, as I found out after a couple of hours of compiling ;)

I hate computers! (Lies, I'm a software engineer)

closed 2020-02-23 by stale[bot]

roboshoes, 2020-11-09:
I am so glad I found this thread. Node version 9 totally solved it for me as well.