serialport/node-serialport #1999: Make node-serialport context-aware

clementAC, 2019-12-16:
I'm not an expert on this. 

I'm using node-serialport in electron. From what I know libraries like node-serialport may be loaded multiple times in different context. 

In the last "LTS" electron version 7.X  there is a deprecation warning concerning non context aware libraries.

Here is the related content I found about this context aware "problem" : 
- https://github.com/electron/electron/issues/18397
- https://nodejs.org/docs/latest-v12.x/api/addons.html#addons_context_aware_addons

Just posting this issue to discuss about the topic and see how I can help despite the fact that i'm not a C/C++ expert

6XGate, 2019-12-20:
It would be great to make this happen so the modules can still be used in the rendering process if no external content is being used.

reconbot, 2019-12-20:
This would help with workers too #1938 

Looking for volunteers.

closed 2019-12-20 by reconbot