serialport/node-serialport #2000: receive data with bad breaking line

a1keke, 2019-12-19:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
- What happens?
- What should have happened?

my question is the same as [#1714 ](https://github.com/serialport/node-serialport/issues/1714)

> Like it was finding some "\n" in the mid of the Serial prints, telling the bus that the write is complete.

if device send the same data, like '123456789',but i will receive like :

- '12345'
- '6789'
or
- '12'
- '3456789'

and the other random situations

that issue is closed but not resolve, plz resolve it.

### Code to Reproduce the Issue

```js
const serial = new serialPort('/dev/ttyUSB0', { baudRate: 115200 });

serial.on('data', data => {
  console.log(data.toString());
})

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
v7.1.5
- Node.js v?
v8.16.1
- Windows? Linux? Mac?
linux, but my application is  in docker
- docker version 
v19.03.4
- Hardware and chipset? (Prolific/FTDI/Other)

closed 2020-02-24 by stale[bot]

mikeyny, 2020-03-01:
reopen, this is a huge issue, which needs to be resolved

reconbot, 2020-03-09:
You need to use a parser. https://serialport.io/docs/api-parsers-overview