serialport/node-serialport #2005: Error with baudRate: Invalid argument setting custom baud rate of 9600

Jmunapo, 2019-12-28:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
Initializing new SerialPort instance
- What happens?
Return an error ```[Error: Error: Invalid argument setting custom baud rate of 9600]```
- What should have happened?
Return new SerialPort
### Code to Reproduce the Issue

```js
const serial = new SerialPort( path, ( err ) => {
    console.log(err);
});

serial.on('data', data => {
  console.log(data.toString());
})
// Code
```

### Versions, Operating System and Hardware

- SerialPort@ 8.0.6
- Node.js v 10
- Linux
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2019-12-29:
What hardware are you trying to use? This is supported by serialport but maybe not your hardware.

Jmunapo, 2019-12-29:
I was trying it on keyboard and mouse but I'm building it for an inverter to monitor power

hackasaur, 2020-02-29:
i'm having the same issue. but it's not just `nodeJS` even when I use `stty` in bash terminal i get a similar error. 
```$ sudo stty -F -a /dev/hidraw1 9600                                                                        1 ↵
stty: invalid argument ‘/dev/hidraw1’
```
when i use serialport in node to read data from the port:
```$ sudo node sensor.js
Error: Invalid argument setting custom baud rate of 9600```

this is the code:
```const SerialPort = require('serialport');
const port = new SerialPort('/dev/hidraw1');

port.on('data', function(data) {
    console.log(data.toString());
});

port.on('error', function(err) {
    console.log(err.message);
  });```

TimLavelle, 2020-03-13:
Good evening... I'm having this same issue as well with a HID device connected to a Raspberry Pi 4

**USB Device:** Zebra LS2208 on /dev/hidraw0
**OS:** Raspbian GNU/Linux 10 (buster)
**Hardware:** Raspberry Pi 4 Model B Rev 1.1 2GB (b03111)

closed 2020-05-20 by stale[bot]