serialport/node-serialport #2006: Error during installation (NW.js)

djipco, 2020-01-02:
I am trying to install the `osc` module which depends on the `serialport` module. Since the project is run inside NW.js, I followed the [appropriate instructions](https://serialport.io/docs/guide-installation#nwjs) and added a `.prebuilrc` file to the root of my project. 

While the installation process does go through all the way to the end, I am getting an error and the `osc` module does not work. In fact, when I try to use `osc`, it reports the following error:

```Cannot find module '@serialport/bindings'```

To find the source of the problem, I tried installing only the `serialport` module and I'm getting the same error during install. 

Ideas?

This is the full log of installation:

```
jpcote@roboter:~/DATA/egodaemon$ npm install serialport

> @serialport/bindings@2.0.8 install /home/jpcote/DATA/egodaemon/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=13.5.0 runtime=node arch=x64 libc= platform=linux)
make: Entering directory '/home/jpcote/DATA/egodaemon/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
../src/serialport.cpp:329:96: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)’
     results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
                                                                                                ^
In file included from /home/jpcote/.cache/node-gyp/13.5.0/include/node/node.h:63:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
../src/serialport.cpp:330:96: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)’
     results->Set(Nan::New<v8::String>("dsr").ToLocalChecked(), Nan::New<v8::Boolean>(data->dsr));
                                                                                                ^
In file included from /home/jpcote/.cache/node-gyp/13.5.0/include/node/node.h:63:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
../src/serialport.cpp:331:96: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)’
     results->Set(Nan::New<v8::String>("dcd").ToLocalChecked(), Nan::New<v8::Boolean>(data->dcd));
                                                                                                ^
In file included from /home/jpcote/.cache/node-gyp/13.5.0/include/node/node.h:63:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/jpcote/.cache/node-gyp/13.5.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
../src/serialport.cpp: In function ‘void EIO_AfterGetBaudRate(uv_work_t*)’:
../src/serialport.cpp:378:106: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::IntegerFactory<v8::Integer>::return_t)’
     results->Set(Nan::New<v8::String>("baudRate").ToLocalChecked(), Nan::New<v8::Integer>(data->baudRate));
[... 51 lines cut ...]
                 from ../src/serialport_unix.cpp:2:
/usr/include/x86_64-linux-gnu/bits/stdio2.h:65:44: note: ‘__builtin___snprintf_chk’ output 22 or more bytes (assuming 1045) into a destination of size 1024
        __bos (__s), __fmt, __va_arg_pack ());
                                            ^
  CXX(target) Release/obj.target/bindings/src/poller.o
  CXX(target) Release/obj.target/bindings/src/serialport_linux.o
  SOLINK_MODULE(target) Release/obj.target/bindings.node
  COPY Release/bindings.node
make: Leaving directory '/home/jpcote/DATA/egodaemon/node_modules/serialport/node_modules/@serialport/bindings/build'

> serialport@8.0.6 postinstall /home/jpcote/DATA/egodaemon/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @serialport/bindings@2.0.8 (node_modules/@serialport/bindings):
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: @serialport/bindings@2.0.8 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: Exit status 1

+ serialport@8.0.6
added 265 packages from 152 contributors and audited 573 packages in 6.075s
found 0 vulnerabilities
```

scytacki, 2020-02-21:
I got a similar error when installing serialport. In my case I was using node.js version 13 running on OS X 10.15.3.  When I downgraded to node.js 12 then serialport installed successfully.

In case it is useful here is one of the 4 compile errors I got:
```
../src/serialport.cpp:329:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
```

I also found this PR in node that fixes their own example:
https://github.com/nodejs/node-addon-examples/pull/121

gmariani, 2020-02-26:
I'm trying to figure out how to get it to work too. I originally had this working on Node 1.2.0, NW.js 0.12.3, and serialport 2.0.6. I'm in the process of trying to update the app and am now using Node 13.8.0, NW.js 0.44.2, and serialport 8.0.7. But when I follow the [instructions](https://serialport.io/docs/guide-installation#nwjs), all it says is to create a .prebuildrc and that's it. When I run the app I get this:
`Uncaught Error: A dynamic link library (DLL) initialization routine failed.
\\?\C:\Users\Mariani\Development\derby-timer\src\node_modules\@serialport\bindings\build\Release\bindings.node
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1246:18)
    at Module.load (internal/modules/cjs/loader.js:1024:32)
    at Function.Module._load (internal/modules/cjs/loader.js:923:14)
    at Module.require (internal/modules/cjs/loader.js:1064:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at bindings (C:\Users\Mariani\Development\derby-timer\src\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (C:\Users\Mariani\Development\derby-timer\src\node_modules\@serialport\bindings\lib\win32.js:1:36)
    at Module._compile (internal/modules/cjs/loader.js:1175:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1201:10)
    at Module.load (internal/modules/cjs/loader.js:1024:32)`

I'm running on a Windows 10 x64 machine. I tried 
`npm install --global node-gyp node-pre-gyp nw-gyp
npm install --global --production windows-build-tools
npm install serialport --build-from-source`
to no avail. I'm in the same boat, what do I need to do to get it to rebuild properly for NW.js?

djipco, 2020-02-26:
As far as `osc` is concerned, the `serialport` module was [made optional](https://github.com/colinbdclark/osc.js-examples/issues/17) which solved my problem.

Having said that, I found it excruciatingly difficult to use any module that rely on the `serialport` module within NW.js. I have had nothing but problems when trying to recompile it. I have had issues with `johnny-five`, `dmx`, `osc`, etc.  At this point, I have pretty much given up... 

If anybody has a solution, I'd love to hear it.

gmariani, 2020-02-26:
I remember having a lot of difficulty with it in the past and when I got it working I never touched it again. Which goes to show why my original program was running serialport v2 and it's now on v8. But it's been long enough, I figured it should be updated and unfortunately all my notes no longer apply. I feel like the documentation is missing a step between creating the .prebuildrc file and actually making it rebuild properly.

scytacki, 2020-02-26:
As I wrote above. I think it only works with Node version 12. Node version 13 seems to have changed some native API that node-serialport is using. I'm not familiar with NW.js, I was installing node-serialport directly. But based on the error reported in the first comment I'd guess it is the same problem.

gmariani, 2020-02-26:
I switched to [NW.js version 0.42.0](https://nwjs.io/blog/v0.42.0/) which is the last version with Node version 12 (12.12.0) and it still displays the same error. I tried running 
```
npm install serialport --build-from-source
```
again to no avail. Any other ideas/suggestions?

djipco, 2020-03-13:
For those interested, I released [`chrome-apps-serialport`](https://github.com/djipco/chrome-apps-serialport) today. Its API matches that of `node-serialport` which means it can be substituted wherever `node-serialport` is used.

I submitted a [PR](https://github.com/firmata/firmata.js/pull/215) to the `firmata.js` team. If they merge it, we will be able to use Johnny-Five within NW.js without having to rely on `node-serialport` at all.

djipco, 2020-03-17:
By using `firmata-io` and my new `chrome-apps-serialport` module, I am now able to use Johnny-Five inside NW.js without having to recompile anything. Here's how:

```js
const SerialPort = require("chrome-apps-serialport").SerialPort;
const Firmata = require("firmata-io")(SerialPort);
const five = require("johnny-five");

SerialPort.list().then(ports => {

  const device = ports.find(port => {
    return port.manufacturer && port.manufacturer.startsWith("Arduino")
  });

  const board = new five.Board({
    io: new Firmata(device.path)
  });

  board.on("ready", () => {
    console.log("Johnny-Five is ready!");
    const led = new five.Led(13);
    led.blink(500);
  });

  board.on("close", () => console.log("Closed!"));

});
```
Cheers!

closed 2020-03-17 by djipco

TamataOceanTeam, 2020-07-31:
> I got a similar error when installing serialport. In my case I was using node.js version 13 running on OS X 10.15.3. When I downgraded to node.js 12 then serialport installed successfully.
> 
> In case it is useful here is one of the 4 compile errors I got:
> 
> ```
> ../src/serialport.cpp:329:14: error: no matching member function for call to 'Set'
>     results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
> ```
> 
> I also found this PR in node that fixes their own example:
> [nodejs/node-addon-examples#121](https://github.com/nodejs/node-addon-examples/pull/121)

Same problem with node v14.7 on Mac Hight Sierra 10.13.6. Dowgrade to node@12 resolved the installation of serialport