serialport/node-serialport #2012: Linux devices whose path begins with /dev/bus/usb aren't presented in list()

Awoogamuffin, 2020-01-09:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?

I'm trying to access a mag-stripe reader on my ubuntu machine through serialport

- What happens?

When running serialport.list() the magstripe reader does not appear.

If I run `udevadm info -e` on the machine, within the huge amounts of data I see this:

`P: /devices/pci0000:00/0000:00:14.0/usb3/3-7
N: bus/usb/003/003
E: BUSNUM=003
E: DEVNAME=/dev/bus/usb/003/003
E: DEVNUM=003
E: DEVPATH=/devices/pci0000:00/0000:00:14.0/usb3/3-7
E: DEVTYPE=usb_device
E: DRIVER=usb
E: ID_BUS=usb
E: ID_MODEL=MagStripe_Card_Reader
E: ID_MODEL_ENC=MagStripe\x20Card\x20Reader
E: ID_MODEL_ID=0009
E: ID_REVISION=0100
E: ID_SERIAL=GIGA-TMS_MagStripe_Card_Reader
E: ID_USB_INTERFACES=:030101:
E: ID_VENDOR=GIGA-TMS
E: ID_VENDOR_ENC=GIGA-TMS
E: ID_VENDOR_FROM_DATABASE=GIGA-TMS INC.
E: ID_VENDOR_ID=1667
E: MAJOR=189
E: MINOR=258
E: PRODUCT=1667/9/100
E: SUBSYSTEM=usb
E: TYPE=0/0/0
E: USEC_INITIALIZED=1747058`

But the item doesn't appear in the list

- What should have happened?

Looking into the linux bindings, it seems like all the information one would need is right here. We have values for DEVNAME for the path, and ID_VENDOR_ENC etc. for the other values, but this item (nor any other item whose devname includes dev/bus/usb) appears in the serialport list.

Now I'm very happy to believe that what this actually means is that I shouldn't be using serialport for this task at all, but rather node-usb. Is that right? I had the opposite problem a while ago when I was trying to use node-usb for something which I should have used serialport for.

Please feel free to call me an idiot if that is the case.

### Code to Reproduce the Issue

```js

//simply listing the data
serialport.list().then(result => {
console.log(result);
}

// Code
```

### Versions, Operating System and Hardware

- SerialPort@?
- Node.js v?
- Windows? Linux? Mac?
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2020-01-10:
It doesn’t appear this is showing up as a serialport with the drivers you
have loaded.

Awoogamuffin, 2020-01-10:
Thanks for replying!

I'm googling to see if there's a way I can have /dev/usb devices appear as serialports, but nothing useful so far. I don't suppose you could point me in the right direction? Is it possible, or even desirable?

It's just that using node-usb to communicate with it directly is confusing me, and there certainly is a kernal driver on the port (I have to detach it from the kernal if I want to claim it in node-usb).

You can see the trouble I'm having here:

https://stackoverflow.com/questions/59673482/having-difficulties-using-node-usb-to-interpret-data-from-a-magstripe-reader

reconbot, 2020-01-10:
Yeah, it's probably pretending to be a keyboard. You can use `process.stdin` to read from the "keyboard" or you can figure out how to get USB keyboard data as text with node-usb.

closed 2020-01-10 by reconbot