serialport/node-serialport #2013: electron-serialport example is broken

gnesher, 2020-01-13:
## 📚 Documentation

The following repository - https://github.com/serialport/electron-serialport which is used to demonstrate the usage of electron-rebuild does not work.

When following instructions and running npm install you get the following error:
'electron-rebuild' is not recognized as an internal or external command,

I've circumvented this by running electron rebuild directly:
./node_modules/.bin/electron-rebuild -f -w yourmodule

No error is thrown, but when running the app (npm run start) I get the following error in the console:

Uncaught Error: The module '\\?\C:\Users\gnesher\work\nortec\test2\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 70. This version of Node.js requires
NODE_MODULE_VERSION 75. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).

SebCanet, 2020-03-12:
Hi,
I have the same problem, example doesn't work, I updated everything an it seems not working anymore.

Uncaught (in promise) TypeError: SerialPort.list no longer takes a callback and only returns a promise
    at Function.SerialPort.list (K:\electron-serialport\node_modules\@serialport\stream\lib\index.js:651)
    at renderer.js:8

I cannot update my projects.

Thanks.

reconbot, 2020-03-12:
That’s a different problem. You can no longer give serialport.list a callback. It returns a promise and you use that to list the ports.

SebCanet, 2020-03-12:
Ok, but example still does not work :-D

reconbot, 2020-03-12:
There are many many issues about electron and how to use electron rebuild. We have some guidance in our docs but since there are so many ways to build and package electron it’s hard to advise you in particular.

SebCanet, 2020-03-12:
Many thanks for your answers. My problem is use serialport to communicate with Arduino board, and documente it to make it easy for anyone else.
I use 'ncu' and last version of Electron, nodjs of course, like any newbie starting on it.
But with a VM with nothing, starting from scratch and trying your example, it does not work.
What can I do ?

closed 2020-05-19 by stale[bot]