serialport/node-serialport #2015: install

Red-Owl, 2020-01-15:
## 👉 [Please follow one of these issue templates](https://github.com/serialport/node-serialport/issues/new/choose) 👈

debian 10 (use a root and not root)
if use -g install success and module serialport not found

prebuild-install WARN install EACCES: permission denied, mkdir '/home/franco/Scrivania/node_modules/@serialport/bindings/build'
gyp WARN EACCES current user ("franco") does not have permission to access the dev dir "/root/.cache/node-gyp/10.15.2"
gyp WARN EACCES attempting to reinstall using temporary dev dir "/home/franco/Scrivania/node_modules/@serialport/bindings/.node-gyp"
gyp WARN install got an error, rolling back install
gyp WARN install got an error, rolling back install
gyp ERR! configure error
gyp ERR! stack Error: EACCES: permission denied, mkdir '/home/franco/Scrivania/node_modules/@serialport/bindings/.node-gyp'
gyp ERR! System Linux 4.19.0-6-amd64
gyp ERR! command "/usr/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/franco/Scrivania/node_modules/@serialport/bindings
gyp ERR! node -v v10.15.2
gyp ERR! node-gyp -v v5.0.5
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/home/franco/Scrivania/package.json'
npm WARN Scrivania No description
npm WARN Scrivania No repository field.
npm WARN Scrivania No README data
npm WARN Scrivania No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@8.0.6 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@8.0.6 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-01-15T18_06_55_626Z-debug.log
root@q4os-desktop:/home/franco/Scrivania#

reconbot, 2020-01-15:
Looks like a local permissions error

Red-Owl, 2020-01-16:
write is enable

> Looks like a local permissions error

on one pc install ok
debian 10 from 9 update

other pc debian 10
not install

please read full.
mkdir fake error report

wrote:
>
access
> the dev dir "/root/.cache/node-gyp/10.15.2"
> gyp WARN EACCES attempting to reinstall using temporary dev dir
> "/home/franco/Scrivania/node_modules/@serialport/bindings/.node-gyp"
> gyp WARN install got [...]
https://github.com/serialport/node-serialport/issues/2015?email_source=notifications&email_token=AAAGK3TUKWDBHLMKDVUQX4DQ55IUNA5CNFSM4KHH6PIKYY3PNVWWK3TUL52HS4DFUVEXG43VMWVGG33NNVSW45C7NFSM4IGNYB2A
https://github.com/

dogatonic, 2020-02-14:
I had a similar problem. My solution: I got more specific about which version of nodejs I was using. I had the latest release (at this writing) 13.8 and I had the same problems as the OP.

So I installed nvm and started using it to pick my nodejs.

    $ nvm install --lts 
    $ nvm use --lts
    Now using node v12.16.0

The current Long Term Support version (at this writing) is 12.6, and after that I had no problems with the serial port. Theoretically, I could have kept nvm'installing different version of nodejs until I got the one that worked.

closed 2020-04-25 by stale[bot]