serialport/node-serialport #2016: Segmentation faults on Raspberry Pi 4 (Alpine Docker container)

cascornelissen, 2020-01-18:
### Summary of Problem
- What are you trying to do?
Using `serialport` to read from a ttyUSB serial port on a Raspberry Pi 4 from within an Alpine Docker container (base image: `node:12-alpine`).

- What happens?
Segmentation fault when connecting

- What should have happened?
No segmentation fault

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport');
const port = new SerialPort('/dev/ttyUSB0', {
    baudRate: 115200
});
```

### Versions, Operating System and Hardware
- SerialPort@8.0.6
- Node.js v12.10.0
- Raspberry Pi 4/Alpine (Docker container)

### Additional information
I'm having a hard time debugging this. A Docker container gets created by copying the source (including `node_modules/` directory) into the directory using `COPY ./ ./`. So my initial though was the problem would be caused by the fact that the dependencies are installed on the Raspberry Pi itself and then copied into the Alpine container.

To fix this I tried to run an extra `RUN npm install --no-save serialport` to ensure the installation of `serialport` is executed within the container context. The strange thing is that this doesn't fix the issue.

When I go into the container manually and run `npm install serialport` (installing `8.0.6` currently) it still segfaults. What's even stranger is that when I run `npm install serialport@7` (installing `7.1.5` currently) it doesn't segfault but the strangest thing is yet to come: running `npm install serialport` (installing `8.0.6` once more) it does **not** segfault anymore.

So I currently have the following line in my Dockerfile to fix it this way:

```
RUN npm install --no-save serialport@7 && npm install --no-save serialport
```

cascornelissen, 2020-03-18:
Although I have found a workaround I would like this to be fixed.

funkyfisch, 2020-04-23:
Are you sure it is not an issue with packages that need c/c++ code compiled?

Try a fresh image instead, and perform the following:
Copy package.json (don't copy package.lock)
Copy ONLY your source files
Run npm install 

Try using the above image again.

What workaround helped you? Maybe there is a hint in there

cascornelissen, 2020-04-23:
You might be correct there. The image gets built on the host system where the container will be running so I was expecting to not run into problems there but since reinstalling it within the container solves the issue that's probably where the problem is.

I don't want to install all dependencies within the container when building because that'd mean I'd have to install all dependencies twice (the Typescript source gets compiled in an earlier step in the pipeline already so it needs dependencies there as well) so I guess my current workaround is the way to go.

The workaround I mentioned is at the bottom of the original post:

```
RUN npm install --no-save serialport@7 && npm install --no-save serialport
```

closed 2020-04-23 by cascornelissen