serialport/node-serialport #2018: Garbage information on a Surface Go

Edig, 2020-01-22:
### Summary of Problem

I connect a weighing machine with a Surface Go via Serial port with a USB-C converter (I install the correct drivers). It works perfectly but, when I start charging the Surface it starts giving me a lot of garbage information and weird characters. I know that the charging port of the Surface it's also a communication Port but I don't know if that's whats interfering or it's my code.

### Code to Reproduce the Issue

```js
function startSerialCon() {
    const SerialPort = require('serialport')
    const Readline = require('@serialport/parser-readline')
    const port = new SerialPort('COM3', { baudRate: 4800 })

    port.on('open', () => console.log('Open connection'))
    port.on('error', error => console.log(error));
    port.on('close', () => console.log('Close connection'))

    const parser = new Readline()
    port.pipe(parser)

    parser.on('data', line => updateWeightNumber(line))
}

function updateWeightNumber(weight) {
    //Remove all garbage data 
    console.log('new line: '+weight)
    console.log('data received: ', weight.toJSON());
    var asciiPattern = /[^\x00-\x7F]/g
    var ascii = weight.match(asciiPattern)
    if (!Array.isArray(ascii)) {
        //Get all numbers and .
        var numberPattern = /\d+.\d/
        console.log(weight)
        value = weight.match( numberPattern )
        console.log(value)
        if (Array.isArray(value) && value.length) {
            value = value.join([])

            num = parseFloat(value)
            actualWeigh = num
            console.log(weight + ' -- ' + num + ' -- '+ value)
            // var printNum = num - tara
            
        }
    }

    calculateNewWeight()
}

// Code
```

### What I tried?

- Change the boundRate to 9,600 or others. It didn't receive any information
- Try to eliminate all garbage data and just get the "Real" data, but its about 99% garbage data
- If I disconnect from charging the surface Go it automatically starts giving me all the correct data. 

### Versions, Operating System and Hardware

- SerialPort@? 8.0.6
- Node.js v? 10.16.0
- Windows? Linux? Mac? Coding in a Mac and dist in Windows
- Hardware and chipset? (Prolific/FTDI/Other) - Surface Go

##Examples of bad and good data
```js
updateWeightNumber("w�00�4.00kg")
    updateWeightNumber("w004.00kg")
```

reconbot, 2020-01-23:
I wonder if there's line noise from the charger that's polluting the data line. This might be a hardware issue and not a software one. To test, maybe put a USB extension cable to bring it father away from the device? If that works, maybe there's shielding or grounding you can employ?

Edig, 2020-01-23:
Another thing I try is. If I turn off the electricity of the outlet and I connect the charger of the surface, it starts giving garbage data, which made me think that it is indeed a line noise, but the outlet is turned off. Maybe changing the COM port to another? 
 
Tomorrow I will try the next:
- Connect the surface to another outlet (different from the weighing machine)
- I will put a USB extension as you suggest and go far away from a line noise it's kind of difficult no?

closed 2020-03-30 by stale[bot]