serialport/node-serialport #2021: Weird console.log behavior from parser

hammooooody, 2020-01-26:
**What are you trying to do?**
Read data coming from serial device.

**What happens?**
I am having a rather weird issue with logging received data from serial.

I have a device that replies with the same data that was sent to it. In this case `abcd` 
When I log the data this is what I get:

```
Received Data >>abcd
abcd
abcd
abcd
...
```
The string `Received Data >>` is completely ignored after the first message.
I tried `console.log('Received Data', data)` as well. Same results!

**What should have happened?**
```
Received Data >>abcd
Received Data >>abcd
Received Data >>abcd
Received Data >>abcd
...
```

### Code to Reproduce the Issue

```js
const SerialPort = require("serialport");
const client = () => {
	const Readline = SerialPort.parsers.Readline;
	const serialPort = new SerialPort("/dev/ttyUSB0");
	const parser = serialPort.pipe(new Readline);
	parser.on("data", data => {
		console.log("Received Data >>" + data);
	});
	setInterval(() => {
		serialPort.write("abcd", err => {
			if (err) console.log(err);
		});
	}, 3000);

};
```

### Versions, Operating System and Hardware

- SerialPort@? 
"serialport": "^8.0.6",
- Node.js v?
v12.14.1
- Windows? Linux? Mac?
Linux
- Hardware and chipset? (Prolific/FTDI/Other)
Raspberry Pi 3b+ 
Arm Cortex A53-architecture

hammooooody, 2020-01-27:
I had to select the proper `delimiter` for the type of data i'm receiving.
say I am getting `{abcd}` I had to change parser to 
```
const parser = serialPort.pipe(new Readline({ delimiter: "}" }));
```
for example 
Everything works!

closed 2020-01-27 by hammooooody

reconbot, 2020-01-27:
🙌

hammooooody, 2020-01-27:
> 🙌

:D 
cheers!