serialport/node-serialport #2025: Error in vscode because of wrong NODE_MODULE_VERSION

maziac, 2020-02-04:
I installed serialport 8.07 on vscode on Mac and got the following error:

Error text when starting the extension:
[...] The module '/Volumes/.../node_modules/@serialport/bindings/build/Release/bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 73. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`)..

My vscode version is:
Version: 1.41.1
Commit: 26076a4de974ead31f97692a0d32f90d735645c0
Date: 2019-12-18T14:57:51.166Z
Electron: 6.1.5
Chrome: 76.0.3809.146
Node.js: 12.4.0
V8: 7.6.303.31-electron.0
OS: Darwin x64 19.2.0

maziac, 2020-04-05:
Hi,
the issues is still there since 2 months.
If this is something that will not be fixed please let me now so I can consider another serial module.
If it will be fixed in the next version please let me know as well and I will wait.

reconbot, 2020-04-05:
This is a problem with visual studio using a different version of node js
than what you installed the package with. Our next version will hopefully
not have this problem but it’s common to all binary packages.

closed 2020-06-12 by stale[bot]