serialport/node-serialport #2026: Download older Version (serialport-v4.0.7-node-v67-linux-arm)?

Klauzen, 2020-02-05:
Hi,
i'm looking for a older version: serialport@4.0.7 (serialport-v4.0.7-node-v67-linux-arm). Is it possible, to download this version?
Thanks!

reconbot, 2020-02-05:
We haven’t had binary prebuilds for arm of any version. The JavaScript is
on npm.

Klauzen, 2020-02-05:
Ok, thanks! I'm a beginner: Could you give me a hint, how i can install version 4.0.7 on a raspberry pi zero w?
`sudo npm install serialport@4.0.7` is not working...

thanks!

reconbot, 2020-02-06:
How isn’t it working? Why do you want the old version?

Klauzen, 2020-02-06:
> How isn’t it working? Why do you want the old version?

I like to use the old version for this git [solar-sis](https://github.com/opengd/solar-sis). It says: 
> Currently it only works with serialport 4.0.7 if you are running with the built in USB to Serial converter on the MPP inverters and chargers.

This happens, when i try to install the 4.0.7:
`pi@raspberrypi:~ $ sudo npm install serialport@4.0.7

> serialport@4.0.7 install /home/pi/node_modules/serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/usr/local/bin/node',
node-pre-gyp verb cli   '/home/pi/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@11.15.0 | linux | arm
node-pre-gyp verb command install []
node-pre-gyp info check checked for "/home/pi/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v67-linux-arm.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v67-linux-arm.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v67-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@11.15.0 (node-v67 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v67-linux-arm.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v67-linux-arm.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@11.15.0 (node-v67 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Connection closed while downloading tarball file
node-pre-gyp verb command build [ 'rebuild' ]
gypgyp WARN  WARNEACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/11.15.0"
 gypEACCES current user ("pi") does not have permission to access the dev dir "/root/.cache/node-gyp/11.15.0"
gyp WARN EACCES  attempting to reinstall using temporary dev dir "/home/pi/node_modules/serialport/.node-gyp"
WARN EACCES attempting to reinstall using temporary dev dir "/home/pi/node_modules/serialport/.node-gyp"
make: Verzeichnis „/home/pi/node_modules/serialport/build“ wird betreten
make: *** Keine Regel vorhanden, um das Ziel „../.node-gyp/11.15.0/include/node/common.gypi“,
  benötigt von „Makefile“, zu erstellen.  Schluss.
make: Verzeichnis „/home/pi/node_modules/serialport/build“ wird verlassen
make: Verzeichnis „/home/pi/node_modules/serialport/build“ wird betreten
make: *** Keine Regel vorhanden, um das Ziel „../.node-gyp/11.15.0/include/node/common.gypi“,
  benötigt von „Makefile“, zu erstellen.  Schluss.
make: Verzeichnis „/home/pi/node_modules/serialport/build“ wird verlassen
gypgyp  ERR! build error
gypERR! build error
 ERR! stack Error: `make` failed with exit code: 2
gyp gypERR!  ERR!stack      at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gypstack Error: `make` failed with exit code: 2
 gypERR!  ERR!stack      at ChildProcess.emit (events.js:193:13)
gypstack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR!  stackERR!     at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
 stack     at ChildProcess.emit (events.js:193:13)
gyp gypERR!  stackERR!      at Process.ChildProcess._handle.onexit (internal/child_process.js:255:12)
System Linux 4.19.75+
gyp gyp ERR!ERR!  System Linux 4.19.75+
command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/Release"
gypgyp ERR!  ERR!cwd  /home/pi/node_modules/serialport
gypcommand "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "build" "--fallback-to-build" "--module=/home/pi/node_modules/serialport/build/Release/serialport.node" "--module_name=serialport" "--module_path=/home/pi/node_modules/serialport/build/Release"
 ERR!gyp  node -vERR! v11.15.0
gyp cwd /home/pi/node_modules/serialport
gyp ERR!  node-gyp -vERR! v5.0.7
 node -vgyp v11.15.0
gyp ERR!  ERR!not ok
 node-gyp -v v5.0.7
gyp ERR! not ok
node-pre-gyp ERR! build error
node-pre-gyp ERR! stack Error: Failed to execute '/usr/local/bin/node /usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js build --fallback-to-build --module=/home/pi/node_modules/serialport/build/Release/serialport.node --module_name=serialport --module_path=/home/pi/node_modules/serialport/build/Release' (1)
node-pre-gyp ERR! stack     at ChildProcess.<anonymous> (/home/pi/node_modules/serialport/node_modules/node-pre-gyp/lib/util/compile.js:83:29)
[... cut ...]
npm WARN pi No README data
npm WARN pi No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-02-06T12_52_09_467Z-debug.log
`

Maybe you can help me? Thanks!

reconbot, 2020-02-06:
Looks like maybe you shouldn’t install it as root? It’s falling back to the pi user in there. If you do need it as root look at our install docs on the —unsafe-perms flag.

I looked at the project, neat! I don’t see any obvious reasons why it couldn’t just be upgraded. That might help too.

Klauzen, 2020-02-06:
Thanks for answering. Hm, even as normal user / flags I can't install the old version.
Now I installed the newest version of serialport again and will ask in the solar-sis threads for specific help - maybe the trouble is about the USB-to-Serial driver of my solar inverter...

But thank you!

closed 2020-04-14 by stale[bot]