serialport/node-serialport #2027: Bindings error

melissapott, 2020-02-06:
### Summary of Problem

I'm trying to read input from USB port in a Vue app.  Just requiring SerialPort causes the following error:
`bindings.js?dfc1:29 Uncaught TypeError: Cannot read property 'modules' of undefined
    at Object.eval (bindings.js?dfc1:29)
    at eval (bindings.js:223)
    at Object../node_modules/bindings/bindings.js (chunk-vendors.js:232)
    at __webpack_require__ (app.js:786)
    at fn (app.js:151)
    at eval (linux.js?88eb:2)
    at Object../node_modules/@serialport/bindings/lib/linux.js (chunk-vendors.js:65)
    at __webpack_require__ (app.js:786)
    at fn (app.js:151)
    at Object.eval (index.js?c888:14)`

### Code to Reproduce the Issue

Spin up the default Vue CLI 'hello world' starter project and add the following to HelloWorld.vue:
```js
const SerialPort = require('serialport')
```
I've also tried defining Serialport as an external module for webpack:
```
module.exports = {
    pluginOptions: {

        electronBuilder: {
            externals: ['serialport']
        }
    }
}
```
### Versions, Operating System and Hardware

- SerialPort@? - 8.0.7
- Node.js v? - 10.13.0
- Windows? Linux? Mac? - Mac
- Hardware and chipset? (Prolific/FTDI/Other)

closed 2020-02-07 by melissapott

mehcode, 2020-02-19:
We're getting the same error on trying to use the `usb` module. Any ideas?

Ludo-Zgt, 2020-05-28:
Why is this issue closed ? how did you resolved it  please ?

reconbot, 2020-05-28:
This is generally a webpack issue as binary modules can’t be webpacked

melissapott, 2020-05-29:
I re-initiated the project using https://github.com/SimulatedGREG/electron-vue, which apparently includes some kind of wizardry, and it worked beautifully.

mumairofficial, 2020-09-21:
@melissapott just for information: https://github.com/SimulatedGREG/electron-vue author is excluding all dependencies by adding them into externals

https://github.com/SimulatedGREG/electron-vue/blob/45a3e224e7bb8fc71909021ccfdcfec0f461f634/template/.electron-vue/webpack.main.config.js#L16

alex-bluetrain, 2021-02-25:
> @melissapott just for information: https://github.com/SimulatedGREG/electron-vue author is excluding all dependencies by adding them into externals

shouldn't that be the same as adding serialport as external in vue.config.js as melissapott did?
```
module.exports = {
  pluginOptions: {
    electronBuilder: {
      externals: ['serialport']
    }
  }
}
```
I'm experiencing the same problem, I might go with that boilerplate from SimulatedGREG, but I rather fix it on my existing project to learn more about it.

any ideas?

Shiva127, 2021-06-04:
> shouldn't that be the same as adding serialport as external in vue.config.js as melissapott did?

No, it's not enough. To make it work you need to ([documentation](https://serialport.io/docs/guide-installation/#invoking-serialport-within-the-renderer-processes)):

1. Set `app.allowRendererProcessReuse = false`.
2. Configure webpack to ignore `import serialport from 'serialport'` and make it available as an external dependency.
3. Set `nodeIntegration = true` & `contextIsolation = false`.

For 2 & 3 use `vue.config.js`:
``` javascript
const { IgnorePlugin } = require('webpack')

module.exports = {
  pluginOptions: {
    plugins: [
      new IgnorePlugin({
        resourceRegExp: /serialport/
      })
    ],
    electronBuilder: {
      nodeIntegration: true,
      externals: [
        'serialport'
      ]
    }
  }
}
```

achyde, 2021-06-08:
I am using electron and also had to add   app.allowRendererProcessReuse = false; before the call to createWindow();

RomeshEtulgama, 2022-06-22:
> > shouldn't that be the same as adding serialport as external in vue.config.js as melissapott did?
> 
> No, it's not enough. To make it work you need to ([documentation](https://serialport.io/docs/guide-installation/#invoking-serialport-within-the-renderer-processes)):
> 
> 1. Set `app.allowRendererProcessReuse = false`.
> 2. Configure webpack to ignore `import serialport from 'serialport'` and make it available as an external dependency.
> 3. Set `nodeIntegration = true` & `contextIsolation = false`.
> 
> For 2 & 3 use `vue.config.js`:
> 
> ```js
> const { IgnorePlugin } = require('webpack')
> 
> module.exports = {
>   pluginOptions: {
>     plugins: [
>       new IgnorePlugin({
>         resourceRegExp: /serialport/
>       })
>     ],
>     electronBuilder: {
>       nodeIntegration: true,
>       externals: [
>         'serialport'
>       ]
>     }
>   }
> }
> ```

This here is the best solutions and the only solution for me. Great!. You deserve my utmost appreciation. Thanks once again.