serialport/node-serialport #2028: parser.on('data', function) still exists after the serial port is closed and the hardware is removed.

junhuanchen, 2020-02-07:
It needs to execute parser.removelistener ('data', deal_device) to remove the serialport before reconnecting after it is found.

Otherwise it will trigger many times.

[image]

[image]

Have a better solution?

reconbot, 2020-02-07:
You could also use a new port object instead of reusing it.

junhuanchen, 2020-02-07:
That's true

junhuanchen, 2020-02-07:
I'd like to discuss it.

Can I have the port removed to notify the parser？

reconbot, 2020-02-08:
This might work https://nodejs.org/api/stream.html#stream_readable_unpipe_destination

junhuanchen, 2020-02-08:
Thanks. I'll try

closed 2020-02-08 by junhuanchen