serialport/node-serialport #2031: electron-v76 support

mkurczewski, 2020-02-12:
### Summary of Problem

Hi. I'm trying to build an electron app but during the `install prebuilt binary` process I'm getting messages like (for each OS)
```bash
prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@8.0.7/bindings-v8.0.7-electron-v76-linux-x64.tar.gz

prebuild-install WARN install No prebuilt binaries found (target=8.0.0 runtime=electron arch=x64 libc= platform=linux)
```

which causes an error
```
../node_modules/app-builder-lib/node_modules/app-builder-bin/mac/app-builder exited with code ERR_ELECTRON_BUILDER_CANNOT_EXECUTE
```

Quick check shows that there's just no release for Electron 8.0.0 as the highest is `v75` but new Electron requires `v76`: https://github.com/serialport/node-serialport/releases

When will you add a support for Electron 8.0.0? It's a stable version now.

### Versions, Operating System and Hardware
- SerialPort 8.0.7
- electron 8.0.0
- electron-builder 22.3.2
- electron-builder-http 19.27.5 
- Node.js v12.9.1
- All platforms

qdimka, 2020-04-03:
@reconbot how we can help you resolve this issue?

ghost602, 2020-04-20:
I set it back from version 8 to 7.
I am using it now.

qdimka, 2020-04-20:
@ghost602 please, try electron-rebuild package. I'm using electron 8.2.0 and It`is working. 
There is hook in package.json.
```
 "scripts": {
    "install": "electron-rebuild -f -w serialport",
  },
```

mkurczewski, 2020-04-21:
> @ghost602 try electron-rebuild package. I'm using electron 8.2.0 and It`is working.
> There is hook in package.json.
> 
> ```
>  "scripts": {
>     "install": "electron-rebuild -f -w serialport",
>   },
> ```

I checked and I still don't believe - it really works 🎉 Rebuilding with node-gyp, electron-rebuild any many other solutions just didn't worked till now. Thanks @qdimka for response! 😄

mkurczewski, 2020-04-29:
Meh... it works only for mac. Linux and windows are having the same issue as mentioned in first post.

panuhorsmalahti, 2020-05-07:
Yeah, I also get this issue (trying to build for Windows on Docker).

reconbot, 2020-05-10:
I feel like this information should make its way into our docs https://github.com/serialport/website/blob/master/docs/guide-installation.md#electron but I don't understand nor can I

reconbot, 2020-05-10:
As for the original issue I've taken @panuhorsmalahti's patch, upgraded nan and our prebuild-install and tweaked our build environment to build on node 12. The result?

[image]

We now have prebuilds for electron ABI v76!

panuhorsmalahti, 2020-05-11:
This fixed the issue for me 👍

mkurczewski, 2020-05-28:
Yay, thanks! It works well 🚀

closed 2020-05-28 by mkurczewski

Booster2ooo, 2020-05-30:
Sorry to bump this up, not sure it is allowed or if I should create a new issue but the same problem appears with Electron ABI v80.

JarrodLodge, 2020-06-02:
Same issue:
`prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ /Users/jarrod/.npm/_prebuilds/950adb-bindings-v9.0.0-electron-v80-darwin-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v80-darwin-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v80-darwin-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.0.1 runtime=electron arch=x64 libc= platform=darwin)`

wuxiaobin1995, 2020-06-13:
rebuilding native dependencies  dependencies=@serialport/bindings@9.0.0 platform=win32 arch=ia32
  • install prebuilt binary  name=@serialport/bindings version=9.0.0 platform=win32 arch=ia32
  • build native dependency from sources  name=@serialport/bindings
                                          version=9.0.0
                                          platform=win32
                                          arch=ia32
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)       
                                          error=prebuild-install info begin Prebuild-install version 5.3.4
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ C:\Users\wxb\AppData\Roaming\npm-cache\_prebuilds\209cc5-bindings-v9.0.0-electron-v76-win32-ia32.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v76-win32-ia32.tar.gz
    prebuild-install WARN install connect ETIMEDOUT 192.30.253.112:443

Booster2ooo, 2020-06-14:
> Same issue:
> 
> ```
> rebuilding native dependencies  dependencies=@serialport/bindings@9.0.0 platform=win32 arch=ia32
>   • install prebuilt binary  name=@serialport/bindings version=9.0.0 platform=win32 arch=ia32
>   • build native dependency from sources  name=@serialport/bindings
>                                           version=9.0.0
>                                           platform=win32
>                                           arch=ia32
>                                           reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)       
>                                           error=prebuild-install info begin Prebuild-install version 5.3.4
>     prebuild-install WARN install prebuilt binaries enforced with --force!
>     prebuild-install WARN install prebuilt binaries may be out of date!
>     prebuild-install info looking for cached prebuild @ C:\Users\wxb\AppData\Roaming\npm-cache\_prebuilds\209cc5-bindings-v9.0.0-electron-v76-win32-ia32.tar.gz
>     prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v76-win32-ia32.tar.gz
>     prebuild-install WARN install connect ETIMEDOUT 192.30.253.112:443
> ```

I think you should double check your internet connexion/proxies because the bindings for v9.0 ABI v76 is available and the link mentionned in the output you provided is working (https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v76-win32-ia32.tar.gz). You had a timeout error (connect ETIMEDOUT 192.30.253.112:443), so I assume it's rather some kind of connectivity issue on your side.

PR #2073 (2020-05-09): Update prebuild to 10.0.0
This hopefully fixes #2031.

As far as I understand, we build the various binaries using the --all parameter with prebuild. This command itself is based on the ABIs in node-abi, which now include electron 8.0.0.

References:
https://github.com/prebuild/prebuild
https://github.com/lgeiger/node-abi#usage