serialport/node-serialport #2044: Ship all native bindings, target at runtime?

busticated, 2020-03-04:
hey there :wave:

thanks for an awesome lib and all the hard work that has gone into maintaining it over the years :pray:

for a number of not-super-great historical reasons, i find myself struggling to distribute a CLI which depends on `serialport`. it's the classic gotcha w/ node-driven CLIs where packaging and distributing them is a pain largely due to having to haul along your own runtime (node.js itself). adding native modules to the mix makes this even trickier (note [pkg](https://github.com/zeit/pkg#native-addons), [nexe](https://github.com/nexe/nexe#native-modules) caveats).

one thing that might help me is simply shipping precompiled (`.node`) native modules for each of the platforms my CLI supports and require'ing the correct one at runtime. theoretically, this at least lets me lean on existing packaging tools which on the whole ignore / don't handle native modules.

it's easy enough to grab the precompile bits from github (e.g. [v8.0.7](https://github.com/serialport/node-serialport/releases/tag/%40serialport%2Fbindings%408.0.7)) but it seems those are only _part_ of what you need in order to replace the `@serialport/Bindings` package..?

short of maintaining a fork with a patch to make [this line](https://github.com/serialport/node-serialport/blob/master/packages/bindings/lib/darwin.js#L2) and similar platform-specific, is there anything i can do given how `@serialport/Bindings` is built / works today?

### Versions, Operating System and Hardware

- SerialPort@8?
- Node.js v12?
- Windows? Linux? Mac? yes
- Hardware and chipset? (Prolific/FTDI/Other) 

thanks in advance 🙏 :+1:

reconbot, 2020-03-05:
Those are everything you need to work cross platform.

busticated, 2020-03-06:
@reconbot 

> Those are everything you need to work cross platform.

sorry, i'm not following. do you mean the `.node` native modules are all i need or..?

reconbot, 2020-03-06:
Yes

busticated, 2020-03-07:
ok, but then i'd be on my own to recreate the contents and functionality of 👉 https://github.com/serialport/node-serialport/tree/master/packages/bindings/lib correct?

reconbot, 2020-03-07:
No, keep all the JS, and it will use [bindings](https://www.npmjs.com/package/bindings) to find the binary module. If you put them all in the paths it looks in, it will find the correct one for the platform.

Mike-Dax, 2020-03-13:
As a side note, we use Yarn PnP and don't like the idea of runtime selection of bindings files, so we have a yarn plugin that rewrites the `bindings` require call to be a static require, essentially outputting this:

```
// ultimately all the bindings package does
module.exports = () => require("./bindings.node")
```

(It also fetches the prebuilds at install time without running the prebuild-scripts.)

https://github.com/Mike-Dax/yarn-prebuilds/blob/master/sources/fetcher.ts#L113

That's our generated code if you want to take a look @busticated

busticated, 2020-03-27:
😬 eeesh what a rough couple of weeks. hope you all are weathering things ok 🙏 

thanks @reconbot and @Mike-Dax for the pointers - turns out the `bindings` module already does exactly what i need 🎉 (hi-five to @TooTallNate once again)

i'll close this up - thanks again 👍

closed 2020-03-27 by busticated