serialport/node-serialport #2048: Error when trying to install (npm install serialport)

Finicnl, 2020-03-07:
### Summary of Problem

(Please answer all 3)

- Trying to install
- Getting errors
- Succesfull install

### Code to Reproduce the Issue

> serialport@4.0.7 install F:\cctalk\node-cctalk-master\node_modules\serialport
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [
node-pre-gyp verb cli   'C:\\Program Files\\nodejs\\node.exe',
node-pre-gyp verb cli   'F:\\cctalk\\node-cctalk-master\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build'
node-pre-gyp verb cli ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@12.16.1 | win32 | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "F:\cctalk\node-cctalk-master\node_modules\serialport\build\Release\serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v72-win32-x64.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v72-win32-x64.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v72-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@12.16.1 (node-v72 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v72-win32-x64.tar.gz
node-pre-gyp verb command build [ 'rebuild' ]
node-pre-gyp ERR! Tried to download(undefined): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v72-win32-x64.tar.gz
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@12.16.1 (node-v72 ABI) (falling back to source compile with node-gyp)
node-pre-gyp http Connection closed while downloading tarball file
node-pre-gyp verb command build [ 'rebuild' ]
Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
Building the projects in this solution one at a time. To enable parallel build, please add the "-m" switch.
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\src\win_delay_load_hook.cc : fatal error C1083: Cannot open compiler generated file: 'F:\cctalk\node-cctalk-master\node_modules\serialport\build\Rele
ase\obj\serialport\win_delay_load_hook.obj': Permission denied [F:\cctalk\node-cctalk-master\node_modules\serialport\build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(90,52): error C2661: 'v8::Value::ToInt32': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\bu
ild\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(94,54): error C2661: 'v8::Value::ToBoolean': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\
build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(98,53): error C2661: 'v8::Value::ToString': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\b
uild\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(102,53): error C2661: 'v8::Value::ToNumber': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\
build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(111,48): error C2661: 'v8::Value::ToString': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\
build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(111,29): error C2512: 'v8::String::Utf8Value::Utf8Value': no appropriate default constructor available [F:\cctalk\node-cctalk-master\node_mo
dules\serialport\build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(118,53): error C2661: 'v8::Value::ToObject': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\
build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(141,100): error C2661: 'v8::Value::ToObject': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport
\build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(174,31): error C2661: 'v8::Value::ToInt32': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\b
uild\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(180,1): warning C4996: 'Nan::Callback::Call': was declared deprecated [F:\cctalk\node-cctalk-master\node_modules\serialport\build\serialport
.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\nan\nan.h(1740): message : see declaration of 'Nan::Callback::Call' [F:\cctalk\node-cctalk-master\node_modules\serialport\build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(193,29): error C2661: 'v8::Value::ToInt32': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\b
uild\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(200,53): error C2661: 'v8::Value::ToObject': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\
build\serialport.vcxproj]
F:\cctalk\node-cctalk-master\node_modules\serialport\src\serialport.cpp(217,116): error C2661: 'v8::Value::ToInt32': no overloaded function takes 0 arguments [F:\cctalk\node-cctalk-master\node_modules\serialport\
[... 255 lines cut ...]
node-pre-gyp ERR! System Windows_NT 10.0.17763
node-pre-gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "F:\\cctalk\\node-cctalk-master\\node_modules\\serialport\\node_modules\\node-pre-gyp\\bin\\node-pre-gyp" "install" "--fallback-to-build"
node-pre-gyp ERR! cwd F:\cctalk\node-cctalk-master\node_modules\serialport
node-pre-gyp ERR! node -v v12.16.1
node-pre-gyp ERR! node-pre-gyp -v v0.6.32
node-pre-gyp ERR! not ok
Failed to execute 'C:\Program Files\nodejs\node.exe C:\Program Files\nodejs\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js build --fallback-to-build --module=F:\cctalk\node-cctalk-master\node_modules\seria
lport\build\Release\serialport.node --module_name=serialport --module_path=F:\cctalk\node-cctalk-master\node_modules\serialport\build\Release' (1)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! serialport@4.0.7 install: `node-pre-gyp install --fallback-to-build`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the serialport@4.0.7 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     C:\Users\hp\AppData\Roaming\npm-cache\_logs\2020-03-07T12_04_28_148Z-debug.log

### Versions, Operating System and Hardware

- SerialPort@ 4.0.7
- Node.js v12.16.1 
- Windows
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2020-03-07:
You need to upgrade serialport for modern nodejs. You’re installing 4x
which only supports up to node v6 if I recall correctly.

Finicnl, 2020-03-07:
Aaaaha thank you for the heads up. I will look it up.

closed 2020-03-07 by Finicnl