serialport/node-serialport #2049: Windows 10, Opening COMx: File not found

brammh, 2020-03-08:
Hi, I would like to read some data from a USB device, and found some code that is doing this for my device using the serialport node on Linux. Want to try this with a W10 machine, but cannot find the right port to do this.

Device manager location: Port_#0003.Hub_#0011
node version: 10.16.0
serialport version: 8.0.7

I was expecting this to be the startpoint to open COM3.
tried it with \\.\ etc

I installed CLI tool, serialport-list which does not list any device.

`var SerialPort = require("serialport");
var port = new SerialPort("COM3");

port.on('open', function() {
  port.write('main screen turn on', function(err) {
    if (err) {
      return console.log('Error on write: ', err.message);
    }
    console.log('message written');
  });
});

port.on('error', function(err) {
  console.log('Error: ', err.message);
})`

code above results in:
Error:  Opening COM3: File not found

I'm sure I'm doing something conceptually very wrong. Can you please guide me in the right direction?
thanks!
bram

reconbot, 2020-03-08:
You can probably find it in device manager? If you do a serial port list
you might find it.

brammh, 2020-03-10:
Thanks Francis,
stupid mistake on my side. Didn't install a proper usb serial driver. I didn't need that on my unix device, so forgot about it on my W10 machine. thanks for helping!
bram

closed 2020-03-10 by brammh