serialport/node-serialport #2051: (node:80839) Electron: Loading non context-aware native modules in the renderer process is deprecated

xenkuo, 2020-03-11:
### Summary of Problem
(node:80839) Electron: Loading non context-aware native modules in the renderer process is deprecated

Will node-serial adapt to be a context-aware native module? As in renderer process communicate to serial port is much more easier than in main process of Electron.
(Please answer all 3)

- What are you trying to do?
I mode a new cross platform serial port tool named comNG. Thanks for the great lib, it works quite well. The only problem is the above warning message.
- What happens?
Electron report above warning message.
- What should have happened?
No this warning message.
### Code to Reproduce the Issue
Use serial port lib in renderer process of electron.
```js
const serialport = new SerialPort(path)

// Code
```

### Versions, Operating System and Hardware
Version: 8.0.7
OS: mac, win, linux
Hardware: normal serialport tool

- SerialPort@?
- Node.js v?
- Windows? Linux? Mac?
- Hardware and chipset? (Prolific/FTDI/Other)

reconbot, 2020-03-11:
Yes, it's something we'd like to do. It's something we need help doing.

https://github.com/serialport/node-serialport/search?q=context-aware+is%3Aopen&unscoped_q=context-aware+is%3Aopen&type=Issues

closed 2020-05-17 by stale[bot]

dario-pedro, 2020-05-31:
In my case I'm getting an error instead of a warning:

`vendor.js:43107 ERROR Error: Loading non-context-aware native module in renderer: '/Users/dpedro/beyond-skylines/angular-electron/node_modules/@serialport/bindings/build/Release/bindings.node', but app.allowRendererProcessReuse is true. See https://github.com/electron/electron/issues/18397.
    at process.func (:4200/electron/js2c/asar.js:140)
    at process.func [as dlopen] (:4200/electron/js2c/asar.js:140)
    at Object.Module._extensions..node (:4200/internal/modules/cjs/loader.js:1016)
    at Object.func (:4200/electron/js2c/asar.js:140)
    at Object.func [as .node] (:4200/electron/js2c/asar.js:140)
    at Module.load (:4200/internal/modules/cjs/loader.js:816)
    at Module._load (:4200/internal/modules/cjs/loader.js:728)
    at Module._load (:4200/electron/js2c/asar.js:717)
    at Function.Module._load (:4200/electron/js2c/asar.js:717)
    at Module.require (:4200/internal/modules/cjs/loader.js:853)`

I'm using the following packages:

`"devDependencies": {
    "@angular-builders/custom-webpack": "9.1.0",
    "@angular-devkit/build-angular": "0.901.4",
    "@angular-eslint/builder": "0.0.1-alpha.18",
    "@angular/cli": "9.1.4",
    "@angular/common": "9.1.4",
    "@angular/compiler": "9.1.4",
    "@angular/compiler-cli": "9.1.4",
    "@angular/core": "9.1.4",
    "@angular/forms": "9.1.4",
    "@angular/language-service": "9.1.4",
    "@angular/platform-browser": "9.1.4",
    "@angular/platform-browser-dynamic": "9.1.4",
    "@angular/router": "9.1.4",
    "@ngx-translate/core": "12.1.2",
    "@ngx-translate/http-loader": "4.0.0",
    "@types/jasmine": "3.5.10",
    "@types/jasminewd2": "2.0.8",
    "@types/mocha": "7.0.2",
    "@types/node": "12.11.1",
    "@typescript-eslint/eslint-plugin": "2.27.0",
    "@typescript-eslint/parser": "2.27.0",
    "chai": "4.2.0",
    "codelyzer": "5.2.2",
    "conventional-changelog-cli": "2.0.32",
    "core-js": "3.6.5",
    "cross-env": "7.0.2",
    "devtron": "1.4.0",
    "electron": "8.2.5",
    "electron-builder": "22.6.0",
    "electron-reload": "1.5.0",
    "eslint": "6.8.0",
    "eslint-plugin-import": "2.20.2",
    "jasmine-core": "3.5.0",
    "jasmine-spec-reporter": "5.0.2",
    "karma": "5.0.4",
    "karma-coverage-istanbul-reporter": "3.0.0",
    "karma-electron": "6.3.0",
    "karma-jasmine": "3.1.1",
    "karma-jasmine-html-reporter": "1.5.3",
    "mocha": "7.1.2",
    "npm-run-all": "4.1.5",
    "rxjs": "6.5.5",
    "spectron": "10.0.1",
    "ts-node": "8.9.1",
    "tslib": "1.11.1",
    "typescript": "3.8.3",
    "wait-on": "4.0.2",
    "webdriver-manager": "12.1.7",
    "zone.js": "0.10.3"
  },
  "engines": {
    "node": ">=10.13.0"
  },
  "dependencies": {
    "electron-rebuild": "1.11.0",
    "serialport": "9.0.0"
  }`

Any further update?

reconbot, 2020-05-31:
nope, nobody has helped yet

michanagel, 2020-06-13:
also getting an error in latest Electron 9.0.4 when serialport is imported into Renderprocess - current workaround is to set `app.allowRendererProcessReuse = false;` in main process, but as Electron has announced this flag will be fully deprecated in the near future.

devrim-oguz, 2020-09-08:
> In my case I'm getting an error instead of a warning:
>
> `vendor.js:43107 ERROR Error: Loading non-context-aware native module in renderer: [...]

Waiting for an update, my Electron build on Raspberry Pi gives the same error...

EngJay, 2020-09-09:
@reconbot I'd consider contributing on this but I'll have to get up-to-speed on the code base and what needs to be done. I've read-up a bit on the issue in the discussions in the electron repo but haven't had a chance to go further than that.

arturbp, 2020-11-13:
Getting the same here when try to use 'usb' lib...

illysky, 2020-11-15:
Waiting here too!

jkozniewski, 2021-09-06:
Well.. old workaround with `app.allowRendererProcessReuse = false;` is not working anymore... 
So there is no way to use serial port in electron apps right now at all ? Is there any viable workaround ?

reconbot, 2021-09-06:
We're actively working on a napi migration which should solve these problems. For now the latest electron may not work.

lucas-sg, 2022-04-10:
> We're actively working on a napi migration which should solve these problems. For now the latest electron may not work.

@reconbot any way I might help with that?

reconbot, 2022-04-11:
Yeah it was released in v10 [blogpost](https://opencollective.com/serialport/updates/serialport-v10-has-been-released) [upgrade guide](https://serialport.io/docs/guide-upgrade#upgrading-from-9x-to-10x)