serialport/node-serialport #2052: Error: Invalid argument setting custom baud rate of 9600

TimLavelle, 2020-03-13:
### Summary of Problem
Good evening... 

- Cannot get a Zebra LS2208 barcode scanner working on /dev/hidraw0
- I get an error telling me of an invalid argument setting custom baud rate
- Serial Port opens
- I have followed the instructions at https://serialport.io/docs/guide-installation#raspberry-pi-linux

### Code to Reproduce the Issue

```js
// Use the serialport module to get scanner  data
const SerialPort = require('serialport');
const port = new SerialPort('/dev/hidraw0',{baudRate: 9600});

// When the port is open just say so
port.on('open', function() {
 console.log('Serial port open');
});

// Check for errors on the serial interface
port.on('error', function(err) {
 console.log(err);
});
```

### Error Output
`[Error: Error: Invalid argument setting custom baud rate of 9600]`

### Versions, Operating System and Hardware
- SerialPort@ 6.9.0 (i installed today via npm, 13 April 2020 - but see Git is saying v is at 8.0.7)
- Node.js v10.16.3
**OS:** Raspbian GNU/Linux 10 (buster)
**Hardware:** Raspberry Pi 4 Model B Rev 1.1 2GB (b03111)
**USB Device:** Zebra LS2208 on /dev/hidraw0

Thanks for any insight!

reconbot, 2020-05-13:
Sorry nobody responded, I would bet the hardware device doesn’t support 9600 or the Pie isn’t setup correctly for a serial port.

closed 2020-05-13 by reconbot

Papouchcom, 2020-09-02:
I also encountered this problem.
The speed on RPi4 could not be set at /dev/ttyS0.
The speed setting on /dev/ttyAMA0 works fine. That's the port I needed, so I didn't deal with it anymore.