serialport/node-serialport #2056: Legacy Electron app: Error trying to download 4.0.7 version

jcramirez, 2020-03-23:
### Summary of Problem

I need to support this legacy app at my work and it uses serialport version 4.0.7. When I run `npm i`

I get the following error:

`node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v79-darwin-x64.tar.gz `

Any ideas why this link is no longer available? 

https://github.com/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v79-darwin-x64.tar.gz

### Versions, Operating System and Hardware

- SerialPort@4.0.7
- Node.js v8.11.2
- Mac

reconbot, 2020-03-23:
https://github.com/serialport/node-serialport/releases/tag/4.0.7

Node 8 didn't exist back then, it also probably doesn't work with 4.0.7.

https://github.com/serialport/node-serialport/blob/4.0.7/README.md#platform-support

jcramirez, 2020-03-26:
> https://github.com/serialport/node-serialport/releases/tag/4.0.7
> 
> Node 8 didn't exist back then, it also probably doesn't work with 4.0.7.
> 
> https://github.com/serialport/node-serialport/blob/4.0.7/README.md#platform-support

I  definelty works with those versions. The problem is if I per example delete the *node_modules* and try to install everything using *npm i* it gives the error I reported. One work around has been to copy the serialport directory inside the node_modules and copy pasting it into other projects. This ONLY works if we use a windows platform.

reconbot, 2020-03-26:
Each version of the nodes Abi has a different release file. Electron and
nodejs have different ones depending on the release. We never had that
file. The node being run with npm doesn’t have to match what electron needs.

closed 2020-06-02 by stale[bot]