serialport/node-serialport #2060: Unable to compile on Node 13(.12.0)

deregtd, 2020-03-28:
I know that 13 isn't LTS yet, but just opening the issue in case you aren't yet aware of it.  We need to switch to Node 13 to work around some issues with ES modules, so this is a bit of a sticking point. :)

Thanks for all of your work on the module!  It works great (outside of this issue) :)

### Summary of Problem

- What are you trying to do?
Install serialport on Node 13.12.0
- What happens?
A compilation error
- What should have happened?
It compiled and installed

### Code to Reproduce the Issue

```
npm i serialport
```

### Versions, Operating System and Hardware

- SerialPort/bindings @ 2.0.8
- Node.js v13.12.0
- Mac OS 10.15.4

### Errors listed
> @serialport/bindings@2.0.8 install /Users/davidderegt/boatkit/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=13.12.0 runtime=node arch=x64 libc= platform=darwin)
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp:329:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(...
    ~~~~~~~~~^~~
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3547:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3550:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/serialport.cpp:330:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("dsr").ToLocalChecked(), Nan::New<v8::Boolean>(...
    ~~~~~~~~~^~~
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3547:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3550:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/serialport.cpp:331:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("dcd").ToLocalChecked(), Nan::New<v8::Boolean>(...
    ~~~~~~~~~^~~
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3547:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3550:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
../src/serialport.cpp:378:14: error: no matching member function for call to 'Set'
    results->Set(Nan::New<v8::String>("baudRate").ToLocalChecked(), Nan::New<v8::Inte...
    ~~~~~~~~~^~~
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3547:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                    ^
/Users/davidderegt/Library/Caches/node-gyp/13.12.0/include/node/v8.h:3550:37: note: 
      candidate function not viable: requires 3 arguments, but 2 were provided
  V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                    ^
4 errors generated.
make: *** [Release/obj.target/bindings/src/serialport.o] Error 1

Raylin51, 2020-03-31:
Try to upgrade `serialport` to 8.0.7 and `@serialport/bindings` to `^8.0.7`.

deregtd, 2020-03-31:
Oh god, I had that in my file, but @canboat brought in 2.0.8 and I didn't notice.  I suck.  I'm so sorry for wasting your time.

closed 2020-03-31 by deregtd