serialport/node-serialport #2061: prebuild-install bindings-v8.0.7 get 404 result

JesseWeb, 2020-03-30:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
trying to build win64 app in macOS
- What happens?

- What should have happened?
get this error
```
 prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@8.0.7/bindings-v8.0.7-electron-v76-linux-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=8.2.0 runtime=electron arch=x64 libc= platform=linux)
```

### Versions, Operating System and Hardware

- SerialPort@8.0.7
- Node.js v10.15.3
- electron@8.2.0
- Mac

mkurczewski, 2020-03-31:
It's a duplicate of #2031 - still unresolved :(

JesseWeb, 2020-04-30:
any progression in this issue?

mkurczewski, 2020-05-28:
@JesseWeb look at https://github.com/serialport/node-serialport/issues/2031. The issue is resolved now.

JesseWeb, 2020-05-28:
> @JesseWeb look at https://github.com/serialport/node-serialport/issues/2031. The issue is resolved now.

Glad to hear that!

closed 2020-08-08 by stale[bot]