serialport/node-serialport #2064: Write does not persist to device if the process exits immediately after the write

joltmode, 2020-04-14:
### Summary of Problem

**What am I doing?**
For deeper context, I asked over on StackOverflow: https://stackoverflow.com/questions/61216551/ftdi-driven-usb-to-rs485-requires-a-undocumented-timeout-after-write

I am developing a controller for Mobitec Bus destination panel. The intended communication layer is `SBC` -> [`USB 2 RS485 [FTDI]`](https://www.metronic.pl/en/products/converters/141-conv485usb) -> `Mobitec`.

While developing, though, my workstation is a Mac OS - if that may change anything.

I have a short-running script, that intends to open, write, wait for successful write and then close the connection. Currently, while prototyping, that's a node process that exits immediately after there's nothing left to do.

P.S. This is my first time doing RS485 and serial interface programming in general. I am definitely unaware of potential nuances.

**What happens?**
I get no errors and I see the activity LED blinking on my USB 2 RS485 device, but it appears that the target device does not receive the message as it does not respond as expected (the display isn't updating the LEDs).

Draining after the write doesn't help either. The write callback and drain both get called at the same time, given that I am doing a single write.

If I was to add a timeout after the write, the updates will eventually come in, assuming the timeout is enough.

**What should have happened?**
I expect that after the write callback or the drain callback, the target device has been updated accordingly.

### Code to Reproduce the Issue

```js
const SerialPort = require("serialport");

const port = new SerialPort("/dev/tty.usbserial-DN05Q09Q", {
  autoOpen: false,
  baudRate: 4800
});

port.on("open", () => {
  console.log("Opened", port.path);
});

port.open((error) => {
  if (error) {
    return void console.error(`Error opening port`, port.path, error);
  }

  port.write(/* Buffer with a length of 23 bytes (proprietary structure) */, (error) => {
    if (error) {
      return void console.error(
        "Error while writing...",
        port.path,
        error
      );
    }

    console.log(
      "Written successfully.",
      port.path
    );
  });
});
```

An added timeout after the write would work though:
```js
port.write(/* Buffer with a length of 33 bytes (proprietary structure) */, (error) => {
  if (error) {
    return void console.error(
      "Error while writing...",
      port.path,
      error
    );
  }

  console.log(
    "Written successfully.",
    port.path
  );

  // ADDED TIMEOUT
  console.log("Starting a timeout after write...");
  setTimeout(() => {
    console.log("Timeout after write expired.");
  }, 1000)
});
```

While I have added a timeout of 1000ms here, I have tested with 10ms and that doesn't work at all, 40ms works ~5% of the times, 50ms seem to be working 99.9% where the remaining 0.01% may be related to other issues.

### Versions, Operating System and Hardware

- SerialPort@?
**8.0.7**
- Node.js v?
**v13.8.0**
- Windows? Linux? Mac?
**MacOS Catalina 10.15.3**
- Hardware and chipset? (Prolific/FTDI/Other)
**FTDI**

---

Is the timeout a known thing in serial world and there are known patterns to work around that? Maybe this is a MacOS issue? Maybe the RS485 adapter requires extra treatment?

Really out of clues here...

fivdi, 2020-04-18:
Does the the device to which the data is being sent reply with a response? If so, can the JavaScript program wait for that response and then terminate?

reconbot, 2020-04-18:
You can also try drain()

joltmode, 2020-04-19:
@fivdi sadly, nope, the RS485 API of that thing is really limited and has write only requests with no acknowledgements.

---

@reconbot as I mentioned in the OP - Draining doesn't help. In my case, drain calls back almost immediately after write (the payload is small).

It's really weird, because the USB to RS485 blinks with it's activity LED in all cases (so the write gets acknowledged on hardware level), but if I exit the Node process while the panel hasn't updated it's LEDs according to latest written state, somehow those messages will not be fully transmitted (even if the RS485 device seems to have received the data).

closed 2020-06-27 by stale[bot]