serialport/node-serialport #2065: Build error nodejs 13.12

splitice, 2020-04-15:
Build error on nodejs 13.12+

```
error /mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings: Command failed.
Exit code: 1
Command: prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild
Arguments:
Directory: /mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings
Output:
prebuild-install WARN install No prebuilt binaries found (target=13.12.0 runtime=node arch=x64 libc= platform=linux)
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@13.12.0 | linux | x64
gyp info find Python using Python version 2.7.17 found at "/usr/bin/python"
gyp info spawn /usr/bin/python
gyp info spawn args [
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/splitice/.cache/node-gyp/13.12.0/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/splitice/.cache/node-gyp/13.12.0',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/splitice/.cache/node-gyp/13.12.0/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
make: Entering directory '/mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings/build'
  CXX(target) Release/obj.target/bindings/src/serialport.o
../src/serialport.cpp: In function ‘void EIO_AfterGet(uv_work_t*)’:
../src/serialport.cpp:329:96: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)’
     results->Set(Nan::New<v8::String>("cts").ToLocalChecked(), Nan::New<v8::Boolean>(data->cts));
                                                                                                ^
In file included from /home/splitice/.cache/node-gyp/13.12.0/include/node/node.h:67:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
../src/serialport.cpp:330:96: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)’
     results->Set(Nan::New<v8::String>("dsr").ToLocalChecked(), Nan::New<v8::Boolean>(data->dsr));
                                                                                                ^
In file included from /home/splitice/.cache/node-gyp/13.12.0/include/node/node.h:67:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
../src/serialport.cpp:331:96: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::FactoryBase<v8::Boolean>::return_t)’
     results->Set(Nan::New<v8::String>("dcd").ToLocalChecked(), Nan::New<v8::Boolean>(data->dcd));
                                                                                                ^
In file included from /home/splitice/.cache/node-gyp/13.12.0/include/node/node.h:67:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
../src/serialport.cpp: In function ‘void EIO_AfterGetBaudRate(uv_work_t*)’:
../src/serialport.cpp:378:106: error: no matching function for call to ‘v8::Object::Set(v8::Local<v8::String>, Nan::imp::IntegerFactory<v8::Integer>::return_t)’
     results->Set(Nan::New<v8::String>("baudRate").ToLocalChecked(), Nan::New<v8::Integer>(data->baudRate));
                                                                                                          ^
In file included from /home/splitice/.cache/node-gyp/13.12.0/include/node/node.h:67:0,
                 from ../../../nan/nan.h:54,
                 from ../src/./serialport.h:6,
                 from ../src/serialport.cpp:1:
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, v8::Local<v8::Value>, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3547:37: note:   candidate expects 3 arguments, 2 provided
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note: candidate: v8::Maybe<bool> v8::Object::Set(v8::Local<v8::Context>, uint32_t, v8::Local<v8::Value>)
   V8_WARN_UNUSED_RESULT Maybe<bool> Set(Local<Context> context, uint32_t index,
                                     ^~~
/home/splitice/.cache/node-gyp/13.12.0/include/node/v8.h:3550:37: note:   candidate expects 3 arguments, 2 provided
bindings.target.mk:114: recipe for target 'Release/obj.target/bindings/src/serialport.o' failed
make: Leaving directory '/mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings/build'
make: *** [Release/obj.target/bindings/src/serialport.o] Error 1
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:275:12)
gyp ERR! System Linux 4.4.0-18362-Microsoft
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /mnt/c/Users/splitice/Documents/GitHub/cc-znp/node_modules/@serialport/bindings
gyp ERR! node -v v13.12.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
```

fivdi, 2020-04-18:
It looks like an old version of serialport is being used. Try with serialport@8.0.7 and it should work.

splitice, 2020-04-23:
Sorry despite not having a package.lock npm was still using the node_modules version. Stange.

closed 2020-04-23 by splitice