serialport/node-serialport #2068: SerialPort hangs while receiving data

mridah, 2020-04-24:
**Version :** serialport ^8.0.7
**Node:** v8.8.0
**OS :** Ubuntu 18.04 LTE 64-bit
**Ram :** 8GB
**Running by:** node app.js     // also tried using  `UV_THREADPOOL_SIZE=20 && node app.js`

I'm using `socat` so create virtual serial ports (mapped), and have created a shell script to write data to the serial port.
I'm connecting to the opposite port using serialport lib, and trying to read/write data.

**Problem :** After reading 1500-2000 lines of data, the program hangs. I'm sure it's the node application that hangs and not the shell script because when I close and restart the node server, it starts reading, and then stops again.

*NodeJS Code*

    const SerialPort = require('serialport');
    const port = new SerialPort('/dev/pts/11', {baudRate: 19200});
    port.on('open', onOpen);

    function onOpen(){
        console.log('Open connections!');
        port.write('heyy')
    }

    function onData(data){
        console.log('on Data ' + data);
        port.write('got data...' + data);   //  <--- I've even tried commenting this
    }

    const parser = new Readline();
    port.pipe(parser);

    port.on('data', onData)
    port.on('error', console.log)
    port.on("close", () => port.open());

*Shell script*

    #!/bin/bash
    for i in {1..1000}
    do
       echo "Welcome $i times"
       echo "$i uhewwe88nnh hbrh3rbhhhu3uy" > /dev/pts/10
    done

This is the output of the shell script (as expected)

[image]

This is the output in nodejs ( it stops after this, i need to restart server )

[image]

After this it stops. No error, nothing. Sometimes the shell scripts also hangs in between, and resumes only when I restart node server. So I guess it's a node issue.

Can you please help me overcome this? @voodootikigod @andrew @reconbot

reconbot, 2020-04-24:
I recommend looking at the debugging output to see what’s going on internally.

https://serialport.io/docs/guide-debugging

mridah, 2020-04-24:
This is the output when I run the shell script first time

[image]

As you can see, it's reading the data correctly here ( on Data 1000 uhewwe88nnh hbrh3rbhhhu3uy is logged)

The following is the output of running the script second time:

[image]

There is no message reading `onData .....` so the data has stopped being read

Does it tell you anything ? @reconbot

theTestTube, 2020-05-12:
**Version :** serialport ^8.0.7, @types/serialport ^8.0.0
**Node:** v12.16.3
**ts-node:** v8.9.1
**OS :** Windows 10
**Ram** : 16GB
**node process assigned RAM at the time it occurs**: ~110MB

I'm afraid we're suffering the same issue, we receive no event to read `on('data')`. In our case we're also testing with a [com0com null modem](https://github.com/Raggles/com0com/releases), but I think this is irrelevant for the issue as the modem works properly when testing processes different of nodejs+serialport.

We've also checked that,
* The `data` event listener is listening when no more messages are read (`port.listenersCount('data')` doesn't decrease after occuring)
* No unexpected exception occurs meanwhile
* Issue occurs with a `Readline` parser, at line number <n>, with <n> being lower as line length increases, after ~2800 lines 7 bytes length

[image]

* Issue also occurs with a `ByteLength` parser

[image]

theTestTube, 2020-05-27:
We've checked this issue occurs only when the serial port has an associated `Parser`, at least with a `Delimiter` or `ByteLength` parser.

If you can parse your port data by simply reading the port with a `port.on("data", function)` and prevent associating a parser with `port.pipe(parse)` the issue doesn't occur.

theTestTube, 2020-05-27:
We've also checked the issue consistently occurs when more than 32768 bytes have been read, independently of buffer size (`highWaterMark`) or buffer reallocation limit (`_kMinPoolSpace`).

[image]

closed 2020-08-02 by stale[bot]

barryvdh, 2020-10-24:
Was this issue ever resolved? My data connection also seems to stall after a certain amount of time.

josegpulido, 2021-03-09:
Any updates? I'm running the same issue as @mridah . The exact debug logs. The app starts with a very high amount of 'bytesToRead'. `serialport/bindings/unixRead read error { [Error: EAGAIN: resource temporarily unavailable, read] errno: -35, code: 'EAGAIN', syscall: 'read' }`. I'm using macOS Mojave 10.14.6, node v10.15.0 and serialport ^9.0.1.

carloscastellanos, 2023-03-17:
Having the same issue.

Stops receiving data after about 30 seconds. All I am doing is reading data from a single analog input of an Arduino.

macOS 12.6.2
Node.js 18.15.0
serialport 10.5.0

Node.js code:

```
const { SerialPort, ReadlineParser } = require("serialport"); // Import the serialport library

let portName = process.argv[2]; // get the port name from the command line

// if they didn't give a port name, tell them so, then quit:
if (!portName) {
  giveInstructions();
}

const myPort = new SerialPort({ path: portName, baudRate: 9600 }); // open the port
const parser = new ReadlineParser(); // make a new parser to read ASCII lines
myPort.pipe(parser); // pipe the serial stream to the parser

// ==== these are the definitions for the serial events: ===== //
myPort.on("open", () => {
  console.log("Serial port opened");
}); // called when the serial port opens

myPort.on("data", (data) => {
  //const val = parseInt(data);
  console.log(data.toString());
}); // Read incoming serial data & print the value to the console

myPort.on("close", () => {
  console.log("port closed.");
}); // called when the serial port closes

myPort.on("error", (err) => {
  console.log(err);
}); // called when there's an error with the serial port

function giveInstructions() {
  //console.log('you did not give a port name');
  console.log("To run this properly, type \n");
  console.log("node serial_input.js \n");
  //console.log('run the listPorts script to get a list of ports.\n');
  process.exit(0);
}
```
Arduino code

```
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0);
  // print out the value you read:
  Serial.print("A0: ");
  Serial.println(sensorValue);
  delay(10);  // delay in between reads for stability
}
```

botsvik, 2023-09-14:
Same issue with just `ReadyParser`. Serialport gets paused after receiving somewhere around 37770 bytes. Without parsers everything works fine.

For now I'm just unpausing the port (`serialport.unpause()`) on every `data` event.

FesterCluck, 2023-09-14:
Antibes having these issues: Node has built in memory limits. You can increase it. However,  a lot of it is your code.  Objects cant just grow forever in a runtime. You need to do some intelligent memory management,  no JS Runtime is perfect. 

Between the two below articles I believe everyone's unanswered questions are covered.  Please read them fully to understand:

https://developer.ibm.com/articles/nodejs-memory-management-in-container-environments

https://blog.appsignal.com/2021/12/08/nodejs-memory-limits-what-you-should-know.html