serialport/node-serialport #2069: when i run "node_modules\.bin\electron-rebuild -v 8.2.3",then i got this errors. help me,thanks vary much

apemen, 2020-04-28:
× Rebuild Failed

An unhandled error occurred inside electron-rebuild
gyp info it worked if it ends with ok
gyp info using node-gyp@6.1.0
gyp info using node@13.1.0 | win32 | x64
gyp info find Python using Python version 2.7.15 found at "C:\Users\Administrator\.windows-build-tools\python27\python.exe"
gyp info find VS using VS2017 (15.9.28307.1093) found at:
gyp info find VS "C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools"       
gyp info find VS run with --verbose for detailed information
gyp info spawn C:\Users\Administrator\.windows-build-tools\python27\python.exe
gyp info spawn args [
gyp info spawn args   'D:\\project\\serialport\\node_modules\\_node-gyp@6.1.0@node-gyp\\gyp\\gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'msvs',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\project\\serialport\\node_modules\\_@serialport_bindings@8.0.7@@serialport\\bindings\\build\\config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'D:\\project\\serialport\\node_modules\\_node-gyp@6.1.0@node-gyp\\addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   'C:\\Users\\Administrator\\.electron-gyp\\8.2.3\\include\\node\\common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=C:\\Users\\Administrator\\.electron-gyp\\8.2.3', 
gyp info spawn args   '-Dnode_gyp_dir=D:\\project\\serialport\\node_modules\\_node-gyp@6.1.0@node-gyp',
gyp info spawn args   '-Dnode_lib_file=C:\\\\Users\\\\Administrator\\\\.electron-gyp\\\\8.2.3\\\\<(target_arch)\\\\node.lib',
gyp info spawn args   '-Dmodule_root_dir=D:\\project\\serialport\\node_modules\\_@serialport_bindings@8.0.7@@serialport\\bindings',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'D:\\project\\serialport\\node_modules\\_@serialport_bindings@8.0.7@@serialport\\bindings\\build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe
gyp info spawn args [
gyp info spawn args   'build/binding.sln',
gyp info spawn args   '/clp:Verbosity=minimal',
gyp info spawn args   '/nologo',
gyp info spawn args   '/p:Configuration=Release;Platform=x64'
gyp info spawn args ]
在此解决方案中一次生成一个项目。若要启用并行生成，请添加“/m”开关。
  serialport.cpp
  serialport_win.cpp
  win_delay_load_hook.cc
c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(8923): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (编译源文件 ..\src\serialport.cpp) 
[D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings\build\bindings.vcxproj]
  c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(7238): note: 参见“v8::MicrotasksCompletedCallback”的声明 (编译源文件 ..\src\serialport.cpp)
c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(8931): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (编译源文件 ..\src\serialport.cpp) 
[D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings\build\bindings.vcxproj]
  c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(7238): note: 参见“v8::MicrotasksCompletedCallback”的声明 (编译源文件 ..\src\serialport.cpp)
c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(8923): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (编译源文件 ..\src\serialport_win.cpp) [D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings\build\bindings.vcxproj]
  c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(7238): note: 参见“v8::MicrotasksCompletedCallback”的声明 (编译源文件 ..\src\serialport_win.cpp)
c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(8931): warning C4996: 'v8::MicrotasksCompletedCallback': Use *WithData version. (编译源文件 ..\src\serialport_win.cpp) [D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings\build\bindings.vcxproj]
  c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(7238): note: 参见“v8::MicrotasksCompletedCallback”的声明 (编译源文件 ..\src\serialport_win.cpp)
C:\\Users\\Administrator\\.electron-gyp\\8.2.3\\x64\\node.lib : fatal error LNK1106: �ļ���Ч���������: �޷����ҵ� 0x54A7C [D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings\build\bindings.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (D:\project\serialport\node_modules\_node-gyp@6.1.0@node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.18363
gyp ERR! command "E:\\nodejs\\node.exe" "D:\\project\\serialport\\node_modules\\_node-gyp@6.1.0@node-gyp\\bin\\node-gyp.js" "rebuild" "--target=8.2.3" "--arch=x64" "--dist-url=https://www.electronjs.org/headers" "--build-from-source"
gyp ERR! cwd D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings
gyp ERR! node -v v13.1.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok

Failed with exit code: 1
[... 53 lines cut ...]
  c:\users\administrator\.electron-gyp\8.2.3\include\node\v8.h(7238): note: 参见“v8::MicrotasksCompletedCallback”的声明 (编译源文件 ..\src\serialport_win.cpp)
C:\\Users\\Administrator\\.electron-gyp\\8.2.3\\x64\\node.lib : fatal error LNK1106: �ļ���Ч���������: �޷����ҵ� 0x54A7C [D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings\build\bindings.vcxproj]
gyp ERR! build error
gyp ERR! stack Error: `C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\MSBuild\15.0\Bin\MSBuild.exe` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onExit (D:\project\serialport\node_modules\_node-gyp@6.1.0@node-gyp\lib\build.js:194:23)
gyp ERR! stack     at ChildProcess.emit (events.js:210:5)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:272:12)
gyp ERR! System Windows_NT 10.0.18363
gyp ERR! command "E:\\nodejs\\node.exe" "D:\\project\\serialport\\node_modules\\_node-gyp@6.1.0@node-gyp\\bin\\node-gyp.js" "rebuild" "--target=8.2.3" "--arch=x64" "--dist-url=https://www.electronjs.org/headers" "--build-from-source"
gyp ERR! cwd D:\project\serialport\node_modules\_@serialport_bindings@8.0.7@@serialport\bindings
gyp ERR! node -v v13.1.0
gyp ERR! node-gyp -v v6.1.0
gyp ERR! not ok

Failed with exit code: 1
    at SafeSubscriber._error (D:\project\serialport\node_modules\_spawn-rx@3.0.0@spawn-rx\lib\src\index.js:267:84)
    at SafeSubscriber.__tryOrUnsub (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.error (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:156:26)
    at Subscriber._error (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:92:26)
    at Subscriber.error (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:72:18)
    at MapSubscriber.Subscriber._error (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:92:26)
    at MapSubscriber.Subscriber.error (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:72:18)
    at SafeSubscriber._next (D:\project\serialport\node_modules\_spawn-rx@3.0.0@spawn-rx\lib\src\index.js:242:65)
    at SafeSubscriber.__tryOrUnsub (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:205:16)
    at SafeSubscriber.next (D:\project\serialport\node_modules\_rxjs@6.5.5@rxjs\internal\Subscriber.js:143:22)

apemen, 2020-04-28:
@joshmh as i use  `yarn install` and `electron-rebuild`  than i got this errors

`App threw an error during load
Error: Cannot find module '@serialport/bindings'
Require stack:
- D:\project\serialport\node_modules\serialport\lib\index.js
- D:\project\serialport\index.js
- E:\nodejs\node_global\node_modules\electron\dist\resources\default_app.asar\main.js   
-
    at Module._resolveFilename (internal/modules/cjs/loader.js:798:15)
c/browser_init.js:7607:16)
    at Module._load (internal/modules/cjs/loader.js:691:27)
    at Module._load (electron/js2c/asar.js:717:26)
    at Function.Module._load (electron/js2c/asar.js:717:26)
    at Module.require (internal/modules/cjs/loader.js:853:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (D:\project\serialport\node_modules\serialport\lib\index.js:2:17)
    at Module._compile (internal/modules/cjs/loader.js:968:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:986:10)`

Huzhengtao, 2020-05-04:
@reconbot pls,I have the same problem, please help me /(ㄒoㄒ)/~~

hanfengcan, 2020-05-04:
make sure you can use serialport in nodejs, then try in electron.

Huzhengtao, 2020-05-06:
It's solved. try different versions of electron， less than 7，it work

reconbot, 2020-05-07:
give serialport@8.0.8 a try

closed 2020-07-13 by stale[bot]