serialport/node-serialport #2070: Read/Write data returns null - Unable to perform any operations

meghajayakumar, 2020-04-28:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
I am trying to establish a read and write connection with an UART device which is a controller of a height adjustment desk.

- What happens?
```
const SerialPort = require('serialport')
const Readline = require('@serialport/parser-readline')
const parser = new Readline()

const port = new SerialPort("COM3",  { 
                baudRate: 9600
 })
```
This is the code I have used and port.read() always returns null value.

For the write operation I have used the code like below:
```
var buf = new Buffer([ 0XF1, 0XF1, 0X01, 0X00, 0X01, 0X7E]);
    port.write(buf, function(err,n) {
    if (err) {
    return console.log('Error on write: ', err.message)
    }
    console.log(n)
    console.log('message written')
})
```
The buffer values are the ones for moving the desk up but no operation takes place and it returns no error or returns undefined value in callback.

More details on the device and setup:
Using an RJ45 to USB connector to connect with the control box of the table.
```
The definition of the SCI is as below:
（Baud Rate）：9600
（Data Mode）：8
（Stop Bit）：1
（Parity Type）：None
Path: COM3
```
[image]

Handset is what is being referred to the laptop.

Basic write operation buffer vals:
Move Up=0XF1 0XF1 0X01 0X00 0X01 0X7E
Move Down=0XF1 0XF1 0X02 0X00 0X02 0X7E
Stop Action=0XF1 0XF1 0X0c 0X00 0X0c 0X7E

Read functionality example:

Current height(1000mm-0x03E8)
0XF2 0XF2 0X01 0X02 0X03 0XE8 0XEE 0X7E 
(there is two bytes in ‘Data’,so ‘Data Length’ is 0x02;‘Checksum’= 0x01 + 0x02 +0x03 +0xE8 = 0xEE)

- What should have happened?
Expecting  the read functionality and write functionality to be able to control the device.

### Versions, Operating System and Hardware

- SerialPort@ ^8.0.7?
- Node.js v10.16.0?
- Windows
- Hardware and chipset? (Prolific/FTDI/Other)
COM3
usb.js:55 FTDIBUS\VID_0403+PID_6001+AD0JIUFZA\0000
usb.js:56 FTDI

danitt, 2020-06-07:
Am attempting the same kind of experiment. @meghajayakumar did you manage to make any progress here?

meghajayakumar, 2020-06-17:
@danitt The read method returns the data only when there is any characteristics changes in the device. For example in my case, it returns the data when the table height is changed. I have not been able to get the data as such yet.