serialport/node-serialport #2075: serialport-list garbled for chinese

flyfishzy, 2020-05-09:
## 💥 Proposal

### What feature you'd like to see
when I use serialport-list in Chinese environment, the returned manufacturer is garbled.
`
C:\Users\Administrator\Desktop\cli>serialport-list -f jsonline

{"path":"COM1","manufacturer":"(��׼�˿�����)","pnpId":"ACPI\\PNP0501\\1"}
{"path":"COM2","manufacturer":"(��׼�˿�����)","pnpId":"ACPI\\PNP0501\\2"}
`
The expected manufacturer is （标准端口类型）

reconbot, 2020-05-09:
I wonder how the registry handles Unicode in c? 🤔

flyfishzy, 2020-05-11:
After reading the code, I tried the following modifications:
`
if (SetupDiGetDeviceRegistryPropertyW(hDevInfo, &deviceInfoData,
                                          SPDRP_MFG, &dwPropertyRegDataType,
                                          reinterpret_cast<BYTE *>(szBuffer),
                                          sizeof(szBuffer), &dwSize))
    {
      int nUtf8Count = WideCharToMultiByte(CP_UTF8, 0, (LPCWCH)szBuffer, -1, NULL, 0, NULL, NULL);
      if (nUtf8Count > 0)
      {
        char *pUtf8Buff = new char[nUtf8Count];
        WideCharToMultiByte(CP_UTF8, 0, (LPCWCH)szBuffer, -1, pUtf8Buff, nUtf8Count, NULL, NULL);
        manufacturer = strdup(pUtf8Buff);
      }
    }
`

Use **SetupDiGetDeviceRegistryPropertyW** to get the unicode string, then convert it to utf-8.
And that seems to work!