serialport/node-serialport #2081: Feature Request: Bundle Typescript types in published package

cinderblock, 2020-05-12:
## 💥 Proposal

### What feature you'd like to see

Typescript types included in the distributed package

## Motivation

It would be nice to not need to install the `@types/serialport` package to use this in TypeScript (or just to get automatic type hinting in modern code editors).

## Pitch

Take the types from https://github.com/DefinitelyTyped/DefinitelyTyped/tree/master/types/serialport and published them in the various packages.

All that is needed is a `package.json#types` field that mirrors the `package.json#main` field. Looks like creating a `types` folder that matches the exported `lib` folder should work.

## Related issues

https://github.com/serialport/node-serialport/issues/1790
[DefinitelyTyped Serialport issues](https://github.com/DefinitelyTyped/DefinitelyTyped/issues?q=is%3Aissue+serialport)

HipsterBrown, 2020-05-12:
It would be possible to make `@types/serialport` a dependency, like [react-testing-library does](https://github.com/testing-library/react-testing-library/blob/master/package.json#L44) for their types. That way they could stay synced with those updates.

cinderblock, 2020-05-13:
https://github.com/DefinitelyTyped/DefinitelyTyped#create-a-new-package
> If you are the library author and your package is written in TypeScript, bundle the autogenerated declaration files in your package instead of publishing to Definitely Typed.

I recognize that this is talking specifically about packages "written in TypeScript", but, frankly, I don't think that is a hard requirement. DefinitelyTyped is wonderful but we don't need to lean on them if we can just bundle the types directly.

cinderblock, 2020-07-12:
Can we get a `never-stale`?

lygstate, 2020-12-17:
bundle it directly, we need @serialport/stream
and so on, especially when using @serialport/stream only and implement custom bindings

karikera, 2021-01-04:
There are too many TypeScript bundlers in npm.
Even I made my own bundler today.

I would be happy if TS supports it directly.

cinderblock, 2021-08-18:
Looking into this a little, it looks like `'serialport'` is using `'plop`` to bundle packages in a consistent way.

Plot claims to support [Typescript types automatically](https://www.npmjs.com/package/plop#typescript-declarations).

Maybe this means what's required is converting everything here to `.ts`?

reconbot, 2021-08-19:
We need a build system in place and then to convert everything to typescript (not just the templates for new parsers)

reconbot, 2022-01-22:
Starting on this with the new bindings for serialport v10 https://github.com/serialport/bindings-cpp/pull/3

reconbot, 2022-01-22:
No promises but if I can make v10 completely in typescript I will

reconbot, 2022-01-23:
`serialport@10.1.0` is using the typscript bindings