serialport/node-serialport #2095: trouble with rebuilding the module for electron 8.1.1

gautamaksh, 2020-06-10:
bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 76.

I am not sure if the problem is with electron or serialport or electron-rebuild

### Versions, Operating System and Hardware

"electron": "^8.1.1",
"electron-rebuild": "^1.11.0",
 "serialport": "^9.0.0"

- Node.js v12.13.0
- Windows 10

MitchellMonaghan, 2020-06-11:
Same issue here tried deleting lock files and node_modules and reinstalling packages. Also tried to use npm to install and npm rebuild which also didn't work.

MitchellMonaghan, 2020-06-11:
After running electron-rebuild my project worked. Figured I would leave this here if it helps you.

gautamaksh, 2020-06-12:
@MitchellMonaghan : Can you tell me which version of node , serialport and electron are you using and on which machine . I have tried many version combinations and failed

MitchellMonaghan, 2020-06-12:
Node v12.16.1
serialport: ^8.0.7
electron: ^8.0.0
electron-rebuild: ^1.8.5

Windows 10