serialport/node-serialport #2099: prebuild-install WARN install prebuilt binaries enforced with --force!

wuxiaobin1995, 2020-06-13:
```
rebuilding native dependencies  dependencies=@serialport/bindings@9.0.0 platform=win32 arch=ia32
  • install prebuilt binary  name=@serialport/bindings version=9.0.0 platform=win32 arch=ia32
  • build native dependency from sources  name=@serialport/bindings
                                          version=9.0.0
                                          platform=win32
                                          arch=ia32
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)       
                                          error=prebuild-install info begin Prebuild-install version 5.3.4
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ C:\Users\wxb\AppData\Roaming\npm-cache\_prebuilds\209cc5-bindings-v9.0.0-electron-v76-win32-ia32.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v76-win32-ia32.tar.gz
    prebuild-install WARN install connect ETIMEDOUT 192.30.253.112:443
```

### Versions, Operating System and Hardware

- SerialPort@9.0.0
- Node.js v 12.18.0
- Windows

reconbot, 2020-06-13:
Looks like it couldn’t connect to the internet?

wuxiaobin1995, 2020-06-14:
Where do I need to put 'bindings-v9.0.0-electron-v76-win32-ia32.tar.gz'，I downloaded it separately.

pakastin, 2020-06-17:
Same issue here..

picklu, 2020-09-04:
Experiencing same issue on my windows 10 for nodejs above 10