serialport/node-serialport #2100: Serialport keeps ESP devices in flash mode

dilin-MS, 2020-06-17:
### Summary of Problem
Hi, our [vscode-arduino](https://github.com/microsoft/vscode-arduino) extension is using the node-serialport package to support device serialport function. Recently we update node-serialport to the latest 8.0.7 and finds the serialport will cause esp devices to be kept in flash mode. See [issue 1015](https://github.com/microsoft/vscode-arduino/issues/1015).

After some investigation, we found a related issue in serialport which has been colsed concerning stale: #1854 . We are encountering the same issue and I didn't see a good solution to this issue. Only a [workaround](https://github.com/serialport/node-serialport/issues/1854#issuecomment-496887057) which suggests the issue might be related to hupcl #1678 . Following the workaround, we fix the issue like this: [fix: serial monitor keeps esp devices in flash mode](https://github.com/microsoft/vscode-arduino/pull/1042/files). This workaround fixes the problem for esp devices, but unforturnately cause new issue: other board's serialport gets trouble. (See [issue 1035](https://github.com/microsoft/vscode-arduino/issues/1053), [issue 1054](https://github.com/microsoft/vscode-arduino/issues/1054))

Therefore, I open this issue to see are there any good fix for fixing the serialport keeping the esp devices in flash mode, so we don't have to use the workaround, which is not a good one since it causes new issues. Thank you very much!

### Versions, Operating System and Hardware

- SerialPort@8.0.7
- Visual Studio Code: 1.43.2
- Windows
- Hardware and chipset? (Prolific/FTDI/Other)
Boards: Wemos D1 Mini, DOIT ESP32 DEVKIT V1

dilin-MS, 2020-08-16:
Are there anyone looking into this issue?

maxpromer, 2020-09-16:
> Are there anyone looking into this issue?

me too

> Are there anyone looking into this issue?

Can fixed by set logic of DTR and RTS line ?

```c++
serialPort.set({
  dtr: true,
  rts: true
});
```

gcampbell-msft, 2023-11-17:
@maxpromer Following up because I'm also hitting this issue in the Serial Monitor extension, and I'm also finding the any call to `set` (depending on the firmware) is causing a reset of the board. See https://github.com/microsoft/vscode-serial-monitor/issues/99.

It'd be great to get an official fix for this.