serialport/node-serialport #2103: Prevent SerialPort from repeating old, outdated responses

petef19, 2020-06-19:
### Summary of Problem

Our Electron app uses Node SerialPort and we write to an attached USB device and then wait for the response via `parser.on('data', ...)`, which uses a ByteLength parser. Sometimes we issues many commands in a row, but each one waits until a response has been received, verifies the response, then moves on to next etc.

For the first 3-4 commands this works fine. But from then on, the  `parser.on('data', ...)` method is repeating **old, outdated responses** that have already been processed in addition to the response of the last command that we're actually waiting for. It would almost appear that SerialPort or Node is caching these old buffers and keeps flushing them out collectively every time a new write command is issued... (?)

We simply only need the response to the last command that we wrote to the port.

I've tried implementing the .flush() and .drain() methods w/ no effect - it is also unclear where exactly these methods need to be called (the docs do not explain much at all here), but I've tried inside the `parser.on('data', ...)`  method (after response has been received) and outside of that method - both did not change behavior.

The only thing that we found that works is issuing `port.unpipe(<parser>)` within `parser.on('data', ...)` after each received response, but I'm not sure whether this is the correct way of doing it (?) and/or if Node or SerialPort are still storing all responses in a buffer somewhere b/c we are not flushing or draining them. There will be times when we will have to issue thousands of commands in a row and we don't want to hog a massive amount of memory.

So, what is the correct, recommended way of flushing/draining these buffers to prevent SerialPort on repeating old, outdated responses every time a new command is being issued ?

### Code to Reproduce the Issue

```
const port = new SerialPort(path);
// write to port
port.write(<cmd>);
// create parser and associate it w/ port
const parser_obj = new this.parser_bl({length: <num_bytes>});
const parser_byte_length = port.pipe(parser_obj);
// parser listener
parser_byte_length.on('data', (data) => {
    // handle response
   ...
   // unpipe parser - comment this out to see SP repeat old responses
   port.unpipe(parser_obj);
});

```

### Versions, Operating System and Hardware

serialport@9.0.0
Node 12.18.0 x64
Electron 8.3.1
Win 10 x64

petef19, 2020-06-24:
can anybody comment on this ?

closed 2020-08-30 by stale[bot]

barryvdh, 2020-09-14:
Did you ever solve this issue? We're encountering the same issue on Raspberry PI 3 and 4. after a random amount of RFID scans from the serialport, the same tag keeps repeating untill we restart the device. Just restarting the Node script does not help, so it seems the buffer is not flushed or something?

reconbot, 2020-09-14:
I'd run in debug mode https://serialport.io/docs/guide-debugging and then open a new issue. This does however sound like a RPI or hardware issue.

petef19, 2020-09-15:
@barryvdh 

yes, we did. 

flushing, draining did not help, but unpiping the parser as demonstrated in my OP above on every on.('data') event does prevent repeating of old, outdated messages...

since nobody commented here on this issue it seems they're not interested in solving this (?)

barryvdh, 2020-09-15:
What does the unpipe do? Do we need to pipe it somewhere again or is it just adding that line in the data callback?

petef19, 2020-09-15:
for the device we are working w/, we get single responses, so unpiping is not an issue as for the next command (that we send to the device) we pipe the parser again.

reconbot, 2020-09-15:
These solutions don't make a lot of sense to me, I'd use debug mode to see if you're continuing to get data.

zhangbowei, 2022-04-22:
Update on this?