serialport/node-serialport #2104: @serialport/bindings are not properly rebuild

petef19, 2020-06-19:
#445 # Summary of Problem

I was looking on options on using serialport in a more "sync" approach, and found [#1758 ](https://github.com/serialport/node-serialport/issues/1758).

Installed latest @serialport/bindings and had electron-rebuild rebuild it for my Electron Node version, but once I open Electron I get these errors:

```
App threw an error during load
Error: The module '\\?\D:\electron\node_modules\@serialport\bindings\build\Release\bindings.node'
was compiled against a different Node.js version using
NODE_MODULE_VERSION 72. This version of Node.js requires
NODE_MODULE_VERSION 80. Please try re-compiling or re-installing
the module (for instance, using `npm rebuild` or `npm install`).
    at process.func [as dlopen] (electron/js2c/asar.js:140:31)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1034:18)
    at Object.func [as .node] (electron/js2c/asar.js:140:31)
    at Module.load (internal/modules/cjs/loader.js:815:32)
    at Module._load (internal/modules/cjs/loader.js:727:14)
    at Function.Module._load (electron/js2c/asar.js:738:28)
    at Module.require (internal/modules/cjs/loader.js:852:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at bindings (D:\electron\node_modules\bindings\bindings.js:112:48)
    at Object.<anonymous> (D:\electron\node_modules\@serialport\bindings\lib\win32.js:1:36)
```

### Code to Reproduce the Issue

package.json
`
{
    "name": "electron",
    "version": "1.0.0",
    "description": "",
    "main": "main.js",
    "scripts": {
        "start": "electron .",
        "install": "electron-rebuild",
        "test": "echo \"Error: no test specified\" && exit 1"
    },
    "author": "",
    "license": "ISC",
    "devDependencies": {
        "serialport": "^9.0.0",
        "@serialport/bindings": "^9.0.0",
        "electron": "9.0.4",
        "electron-devtools-installer": "^3.0.0",
        "electron-rebuild": "^1.11.0"
    }
}
`

main.js
`
const {app, BrowserWindow} = require('electron');

const Binding = require('@serialport/bindings');

function createWindow()
{
    // Create the browser window.
    let win = new BrowserWindow({
        width: 800,
        height: 600,
        webPreferences: {
            nodeIntegration: true
        }
    });

    win.webContents.openDevTools({});

// and load the index.html of the app.
    win.loadFile('index.html');
}

app.whenReady().then(createWindow);
`

run Electron
`npm start`

### Versions, Operating System and Hardware

"serialport": "^9.0.0",
"@serialport/bindings": "^9.0.0",
"electron": "9.0.4",
Node 12.18.0 x64
Win 10 x64

SaikatCool, 2020-06-19:
@petef19 I also have a similar issue https://github.com/serialport/node-serialport/issues/2105.

It works with "electron rebuild" and "electron ." command, but does not work when I use electron-builder for packaging the app.

Let me know, if you manage to find a solution.

petef19, 2020-06-19:
> @petef19 I also have a similar issue #2105.
> 
> It works with "electron rebuild" and "electron ." command, but does not work when I use electron-builder for packaging the app.
> 
> Let me know, if you manage to find a solution.

@SaikatCool: my problem is w/ rebuilding `@serialport/bindings` package, not the general `serialport` package as that rebuilds just fine here using `electron-rebuild`.

So the commands you posted are for `@serialport/bindings` package ? Both of them don't work but I did not install the packages globally, just as dev dependencies...

Lmk how to execute these and I'll give it a try...

xxseabxx, 2020-07-04:
@petef19  got the same problem. any news on that?
for me it seams that the package is also not existing within the release folder.

xxseabxx, 2020-07-04:
@petef19  got the same problem - any update on that.
For mit it seams that in the ticket #2105 the whole v80 is missing.

SaikatCool, 2020-07-04:
I downgraded Electron 9.0.0 to 8.2.5 to solve it.
The ABI for Electron 9.0.0 is not yet released by serialport.

closed 2020-09-11 by stale[bot]