serialport/node-serialport #2105: SerialPort v9.0.0 with electron

SaikatCool, 2020-06-19:
### Summary of Problem
Not able to build Electron app with SerialPort v9.0.0

(Please answer all 3)

- What are you trying to do?
I am using serialport v9.0.0 in my Electron app and building via electron-builder.

"electron": "^9.0.3",
"electron-builder": "22.7.0",
"serialport": "^9.0.0"

- What happens?
When I try to build electron app for windows, it throws this error:
• build native dependency from sources  name=@serialport/bindings
                                          version=9.0.0
                                          platform=win32
                                          arch=x64
                                          reason=prebuild-install failed with error (run with env DEBUG=electron-builder to get more information)
                                          error=prebuild-install info begin Prebuild-install version 5.3.4
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ C:\Users\Z00303MF\AppData\Roaming\npm-cache\_prebuilds\90024b-bindings-v9.0.0-electron-v80-win32-x64.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v80-win32-x64.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.0/bindings-v9.0.0-electron-v80-win32-x64.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.0.3 runtime=electron arch=x64 libc= platform=win32)

  ⨯ cannot execute  cause=exit status 1
                    out=
    > @serialport/bindings@9.0.0 install D:\FutureGrid\charger-config-frontend\node_modules\@serialport\bindings
    > prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

    D:\FutureGrid\charger-config-frontend\node_modules\@serialport\bindings>if not defined npm_config_node_gyp (node "C:\Users\Z00303MF\AppData\Roaming\npm\node_modules\npm\node_modules\npm-lifecycle\node-gyp-bin\\..\..\node_modules\node-gyp\bin\node-gyp.js" rebuild )  else (node "C:\Users\Z00303MF\AppData\Roaming\npm\node_modules\npm\node_modules\node-gyp\bin\node-gyp.js" rebuild )

                    errorOut=prebuild-install WARN install No prebuilt binaries found (target=9.0.3 runtime=electron arch=x64 libc= platform=win32)
    gyp ERR! configure error
    gyp ERR! stack Error: Command failed: C:\Python38\python.EXE -c import sys; print "%s.%s.%s" % sys.version_info[:3];
    gyp ERR! stack   File "<string>", line 1
    gyp ERR! stack     import sys; print "%s.%s.%s" % sys.version_info[:3];
    gyp ERR! stack                       ^
    gyp ERR! stack SyntaxError: invalid syntax
    gyp ERR! stack
    gyp ERR! stack     at ChildProcess.exithandler (child_process.js:303:12)
    gyp ERR! stack     at ChildProcess.emit (events.js:310:20)
    gyp ERR! stack     at maybeClose (internal/child_process.js:1021:16)
    gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
    gyp ERR! System Windows_NT 10.0.17134
    gyp ERR! command "C:\\Program Files\\nodejs\\node.exe" "C:\\Users\\Z00303MF\\AppData\\Roaming\\npm\\node_modules\\npm\\node_modules\\node-gyp\\bin\\node-gyp.js" "rebuild"
    gyp ERR! cwd D:\FutureGrid\charger-config-frontend\node_modules\@serialport\bindings
    gyp ERR! node -v v12.16.3
    gyp ERR! node-gyp -v v3.8.0
    gyp ERR! not ok
    npm ERR! code ELIFECYCLE
    npm ERR! errno 1
    npm ERR! @serialport/bindings@9.0.0 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
    npm ERR! Exit status 1
    npm ERR!
    npm ERR! Failed at the @serialport/bindings@9.0.0 install script.
    npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

- What should have happened?

### Versions, Operating System and Hardware

- SerialPort@ - 9.0.0
- Node.js v?  - v12.16.3
- Windows? Linux? Mac? - Windows 10

SaikatCool, 2020-06-19:
@reconbot  kindly help!

ruudboon, 2020-07-13:
Seeing this as well.

PR #2122 (2020-08-08): Add Electron 9 support
This adds support for prebuilding for Electron 9 by updating to the latest version of node-abi, which is a dependency of both prebuild and prebuild-install.

Support for Electron 8 was added in #2073, by only updating `prebuild`. However, no new `prebuild` version has been released since then, so updating via that route isn't open. That's why it's important to explicitly specify a `node-abi` version, so you can bump that separately as soon as a new Node or Electron version comes out.

Closes #2105
closed 2020-08-08 by reconbot