serialport/node-serialport #2107: Memory leak on Raspberry Pi4 with Raspbian

Fuzzzzel, 2020-07-02:
### Summary of Problem

- What are you trying to do?
I try to use Serialport to connect a port on a Raspberry Pi4 with Raspbian installed. We have used the same setup before on a Raspberry Pi3 with this setup and it was working perfectly fine. Changing to Raspberry Pi4 results in the problem described below.

- What happens?
The result is, that the memory usage rises 1-2MB per second until it is full and the application crashes.

This problem does not occur with the following setups:
Raspberry Pi3 + Raspbian
Raspberry Pi4 + Ubuntu Server
Raspberry Pi3 + Ubuntu Server

- What should have happened?
As in the other setups there should be no additional memory consumption so the Serialport library can be used in a stable setup.

--> Any suggestions on what to change and/or is this a known issue that I am just not able to find in the recent issues?

### Code to Reproduce the Issue
--> For this code no device needs to be connected to the Pi.

```
const SerialPort = require('serialport')
const path = '/dev/ttyAMA0'
const port = new SerialPort(path, { autoOpen: false })
port.open(function (err) {
        if (err) {
                console.log(err)
        } else {
                port.write('TESTDATA', (err) => {
                        console.log(err)
                })
        }
})

// Start reading from stdin so we don't exit.
console.log('--- Press key to continue ---')
process.stdin.resume()
```

### Versions, Operating System and Hardware

- SerialPort@9.0.0 (same issue on Serialport@6.2.2
- Node.js v 10.19.0
- Linux Raspbian <-- not happening on Ubuntu Server
- Raspberry Pi 4  <-- not happening on Raspberry Pi 3

mawi-cs, 2020-07-02:
I have just tested this with Node 14.5.0 and Serialport 9.0.0 on my RPi 4 with the newest Raspi OS as it is called now, and i am getting the same results.

fivdi, 2020-07-06:
When using Raspberry Pi OS on a Raspberry Pi 4, `/dev/ttyAMA0` is the default Bluetooth UART. By default `/dev/serial1` is a link to `/dev/ttyAMA0` so it's the same device.

Running the following command from the command line does the same thing as the test code in the first post above:
```
echo TESTDATA > /dev/ttyAMA0
```
Running this command from the command line also results in issues. It doesn't result in memory leaks but it does result in the corresponding `bash` process blocking one of the 4 CPUs on the Pi 4 requiring 100% CPU.

Do you really want to use the Bluetooth UART and not the UART broken out to pins 8 and 10 on the Raspberry Pi GPIO header?

If the test program is modified to use `/dev/ttyS0` (or `/dev/serial0`) then the test program will function as expected and the UART broken out to pins 8 and 10 on the Raspberry Pi GPIO header will transmit the string TESTDATA successfully.

Fuzzzzel, 2020-07-07:
Thank you very much for pointing this out. In fact deactivating Blutooth solves the issue for us. But with your input we might just check that we do not connect to /dev/ttyAMA0 or /dev/serial1.

Is there anything one can find out if a port is UART like a property that can be read after opening the port? Then I could just not test the port (e.g. not send data to it) but skip it. In this case the solution would be more generic when trying to avoid such a situation.

fivdi, 2020-07-07:
> Is there anything one can find out if a port is UART like a property that can be read after opening the port?

To the best of my knowledge all serialports are UARTs. Bluetooth uses a UART and the serialport broken out to pins 8 and 10 on the GPIO header is a UART. I don't know if node-serialport can be used to determine if a serialport on a Raspberry Pi is for Bluetooth. Perhaps there is information related to the topic in the data returned by `SerialPort.list()`.

TobyBorland, 2020-09-05:
I can verify that a Serialport stream on all serial ports on a Raspberry Pi 4 running the August 2020 Raspbian Buster will result in a  rapid memory leak if bluetooth is enabled at boot.

closed 2020-11-15 by stale[bot]

TiMESPLiNTER, 2022-12-22:
This problem still persists as of today. Disabling Bluetooth through the `/boot/config.txt` fixes the issue.

My RPi: 3b
My OS:

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

SerialPort version: 10.5.0
Node version: 18

Would at least be interesting why exactly this happens.