serialport/node-serialport #2108: electron vue  running Cannot read property 'indexOf' of undefined

396784797, 2020-07-04:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
 serialport Perform error
- What happens?
vue-router.esm.js?8c4f:2171 TypeError: Cannot read property 'indexOf' of undefined
    at Function.getFileName (bindings.js?dfc1:178)
    at bindings (bindings.js?dfc1:82)
    at eval (win32.js?7cf7:1)
    at Object../node_modules/@serialport/bindings/lib/win32.js (about.js:120)
    at __webpack_require__ (index.js:854)
    at fn (index.js:151)
    at eval (index.js?c888:6)
    at Object../node_modules/@serialport/bindings/lib/index.js (about.js:32)
    at __webpack_require__ (index.js:854)
    at fn (index.js:151)
- What should have happened?

### Code to Reproduce the Issue

```js
const serialport = new SerialPort(path)

// Code
```

### Versions, Operating System and Hardware
"vue-cli-plugin-electron-builder": "~2.0.0-rc.1",
- SerialPort@?
"serialport": "^9.0.0",
- Node.js v?
12.17.0
- Windows? Linux? Mac?
Windows
- Hardware and chipset? (Prolific/FTDI/Other)

ronger-x, 2020-07-22:
@396784797 
Hope this helps you 
[indexOf Error with Quasar, Electron, and Serialport #2089](https://github.com/serialport/node-serialport/issues/2089#issuecomment-662570117)

closed 2020-09-27 by stale[bot]

ksmylmz, 2022-01-23:
I use serial port on UI layer (in my case Vue view) and my build target win32

i was including like below as shown documentation usage page. 

const SerialPort = require('serialport).

after long digging i realized that   following line is not pointing to the right path on my computer

const binding = require('bindings')('bindings.node')
(node_modules\@serialport\bindings\lib\win32.js)

when i change like that it worked 
const binding = require('../bin/win32-x64-89/bindings.node')

after i found right importing usage for me that shown bindings page on documentation
[https://serialport.io/docs/api-bindings](url)

it is work for me usage like that,

const SerialPort = require('@serialport/stream')
SerialPort.parsers = {
  Readline: require('@serialport/parser-readline')
}