serialport/node-serialport #2109: RPi / Arm Compile / Installation doc please

jet-leeroy, 2020-07-10:
## 📚 Documentation

Trying to get serialport running for ARM is not clearly documented / or lacking detail. Please can we have a proper step by step from first node install to running example.js / hello.js example. Have spent days trying to get serialport running with node v14.5.0 on Arm7, and always get version mismatch problems. Ultimately I downgraded my project to node 10.19 which seemed to work, but all manner of rebuilds with all number of flags completely failed to bring serialport up to usable with 14.5

### Have you read the [Contributing Guidelines on docs](https://github.com/serialport/node-serialport/blob/master/CONTRIBUTING.md#writing-documentation)?

Yes

reconbot, 2020-08-09:
If your RPI is setup for serial than it shouldn't be any different than building on any other system. What is the error exactly?

mikelyndersOKCC, 2020-09-21:
Hi there, I had no problems installing the latest serialport on my brand new RPi4 with latest nodejs and and Raspberry Pi OS. Maybe try upgrading one or all?

SerialPort@9.0.1
Node.js v14.11.0
Linux (Raspberry Pi 4) Raspberry Pi OS August 2020

LongLiveCHIEF, 2020-09-26:
@jet-leeroy what OS are you using. Is python2 on the PATH for the user running the install?

jet-leeroy, 2020-09-27:
Hey

Sorry, have been away due to work and covid.

The issue is not that it doesn't work, the issue is the documentation
around which versions work with which platforms is light.

I am using a rpi 2 with Ubuntu. First time build spat up a while bunch of
errors probably due to the compiled C native binary being used in the
background for the actual port interface. Eventually dropping my node
version to 12 and rebuilding got it to work, but trying to start with node
14 was a bust. Additionally I had a few build options I needed to use that
took a lot of googling.. I recall some kind of allow unsafe flag I needed
to use at one point.

I'm not saying it doesn't work, I am not a node js dev, my background is
dotnet, so. It took a lot of google to get a a hello world going so I could
move forward.I'm just saying the documentation to get a hello world up on
rpi could've been better. Not everyone is a node dev.

Justin

LongLiveCHIEF, 2020-09-27:
I don't even think this is a node dev thing, I think it's a sysadmin thing. 

I read https://serialport.io/docs/guide-installation#compilation-problems early on, and that's how I figured it out. I've had many, _many_ years as both a developer and sysadmin, so I knew how to interpret those instructions to Ubuntu on the RPi (installed `build-essential` and  `python2`, then ran `update-alternatives` so python2 was the default).

With python 2 being no longer installed in many distros by default, _including_ windows, I agree it may be time to tweak the docs a bit.

[node-gyp](https://www.npmjs.com/package/node-gyp) also now states that python 3.8 is supported (looks like it was added back in March), although I haven't tested it personally. Maybe instead of this being a documentation issue specific to RPi, it's a documentation issue to update compilation problems document to reflect modern OS/toolchain states?

reconbot, 2020-09-27:
PRs welcome 🙏
-- 
---
Francis Gulotta
wizard@roborooter.com

LongLiveCHIEF, 2020-12-06:
go away stale bot

GazHank, 2021-07-05:
The docs have been updated to reflect that Python 3 is now required.  Are there any other updates to the docs required to address this?

LongLiveCHIEF, 2021-07-05:
iirc the problem was that _python 2_ was required. Haven't tested this in a year though, but I can try again in the next day or two.