serialport/node-serialport #2110: Serialport Not reading Any Datas

huangxuewu, 2020-07-11:
### Summary of Problem

- What are you trying to do?
I'm trying to read data from COM2 ( It is a weight scale. Model Magellan 8500)

- What happens?
Not triggering data event

- What should have happened?
Should triggering data
### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')
const port = new SerialPort('COM2');

port.on('data', console.log);
```

The following code is working on python. serial is provided by pyserial
```python
import serial
port = serial.Serial("COM2")

while True:
     out=str(port.read())
     print(out)
```

### Versions, Operating System and Hardware

- SerialPort@? 9.0.0
- Node.js v? 13.14.1
- Windows? Linux? Mac? Windows 10
- Hardware and chipset? (Prolific/FTDI/Other)

The serialport is not reading anything. Any help is appreciated!

huangxuewu, 2020-07-17:
No worries, I figured it out, for people who have the issue like me try to read com port but no 'data' event trigger should try to set dsr to true.

```js
const SerialPort = require('serialport');
const port = new SerialPort('COM2');

port.on('open', () => port.set({ dsr: true }))
port.on('data', (data) =>  console.log("data event:", data));
```

closed 2020-07-17 by huangxuewu

Fusseldieb, 2021-08-05:
Well, I had a problem that my `.on('data')` was firing only ONCE, at the very beginning.

Turns out I forgot to send the `\n` at the end of my command so it gets intepreted by the device. I feel dumb. Wasted 3 hours on this.

midori0507, 2022-09-14:
@huangxuewu thank you, you really save my day