serialport/node-serialport #2111: Error: Cannot find module '@serialport/bindings'

jag100UK, 2020-07-22:
After restarting my machine, I'm having problems with Node Red running on Docker in Ubuntu 20.04 using an npm package that relies on Serialport

From what I understand, the npm named node-red-contrib-rfxcom@2.11.1 has a dependancy on serialport.

When Node Red starts, the following messages appear. in the log

> serialport@9.0.0 postinstall /opt/node_modules/serialport
> node thank-you.js

Thank you for using serialport!
If you rely on this package, please consider supporting our open collective:
> https://opencollective.com/serialport/donate

+ node-red-contrib-rfxcom@2.11.1
added 5 packages from 4 contributors and audited 1074 packages in 4.707s

37 packages are looking for funding
  run `npm fund` for details

found 4 low severity vulnerabilities
  run `npm audit fix` to fix them, or `npm audit` for details
[cont-init.d] user.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[13:42:08] INFO: Starting Node-RED...

> addon-node-red@ start /opt
> node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js"

22 Jul 13:42:08 - [info] 

Welcome to Node-RED
===================

22 Jul 13:42:08 - [info] Node-RED version: v1.1.0
22 Jul 13:42:08 - [info] Node.js  version: v12.17.0
22 Jul 13:42:08 - [info] Linux 5.4.0-40-generic x64 LE
22 Jul 13:42:09 - [info] Loading palette nodes
22 Jul 13:42:10 - [info] Dashboard version 2.22.1 started at /endpoint/ui
/bin/sh: /opt/node_modules/node-red-node-pi-gpio/testgpio.py: not found
22 Jul 13:42:10 - [warn] rpi-gpio : Raspberry Pi specific node set inactive
22 Jul 13:42:10 - [warn] ------------------------------------------------------
**22 Jul 13:42:10 - [warn] [node-red-contrib-rfxcom/rfxcom] Error: Cannot find module '@serialport/bindings'**
Require stack:
- /opt/node_modules/serialport/lib/index.js
- /opt/node_modules/rfxcom/lib/rfxcom.js
- /opt/node_modules/rfxcom/lib/index.js
- /opt/node_modules/rfxcom/index.js
- /opt/node_modules/node-red-contrib-rfxcom/rfxcom.js
- /opt/node_modules/@node-red/registry/lib/loader.js
- /opt/node_modules/@node-red/registry/lib/index.js
- /opt/node_modules/@node-red/runtime/lib/nodes/index.js
- /opt/node_modules/@node-red/runtime/lib/index.js
- /opt/node_modules/node-red/lib/red.js
- /opt/node_modules/node-red/red.js

node-red-contrib-rfxcom@2.11.1
Error: Cannot find module '@serialport/bindings'

If I run "npm install serialport@latest" within the console of the docker container, the following appears

root@a0d7b954-nodered:/root/.npm/_logs$ npm install serialport@latest

> @serialport/bindings@9.0.0 install /root/.npm/_logs/node_modules/@serialport/bindings
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild

prebuild-install WARN install No prebuilt binaries found (target=12.17.0 runtime=node arch=x64 libc=musl platform=linux)
gyp ERR! build error
gyp ERR! stack Error: not found: make
gyp ERR! stack     at getNotFoundError (/usr/lib/node_modules/npm/node_modules/which/which.js:13:12)
gyp ERR! stack     at F (/usr/lib/node_modules/npm/node_modules/which/which.js:68:19)
gyp ERR! stack     at E (/usr/lib/node_modules/npm/node_modules/which/which.js:80:29)
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/which/which.js:89:16
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/index.js:42:5
gyp ERR! stack     at /usr/lib/node_modules/npm/node_modules/isexe/mode.js:8:5
gyp ERR! stack     at FSReqCallback.oncomplete (fs.js:167:21)
gyp ERR! System Linux 5.4.0-40-generic
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /root/.npm/_logs/node_modules/@serialport/bindings
gyp ERR! node -v v12.17.0
gyp ERR! node-gyp -v v5.1.0
gyp ERR! not ok
npm WARN enoent ENOENT: no such file or directory, open '/root/.npm/_logs/package.json'
npm WARN _logs No description
npm WARN _logs No repository field.
npm WARN _logs No README data
npm WARN _logs No license field.

npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! @serialport/bindings@9.0.0 install: `prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the @serialport/bindings@9.0.0 install script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-07-22T17_13_14_247Z-debug.log

Do you have any guidance on how I can fix this?

maxwellhadley, 2020-07-23:
Please see also [https://github.com/maxwellhadley/node-red-contrib-rfxcom/issues/99](url)

fivdi, 2020-07-24:
>  gyp ERR! stack Error: not found: make

It looks like the `make` utility is not installed on the Raspberry Pi. In order to build serialport from source code a number of packages need to be installed. If I remember correctly these packages are `python`, `g++` and `make`.

jag100UK, 2020-07-24:
Hi fivdi

I'm running on an Intel NUC with Ubuntu 20.04 LTS and Docker
I'm not sure why Raspberry Pi is mentioned in the logs, its always been there even when Node Red was working correctly.
Last night I did a fresh install of the Docker Image and the same messages appeared.

I also tried specifying "make" in the config file along with some other modules which have been recommended on the Home Assistant forum, but this hasn't worked either

```
system_packages:
  - make
npm_packages:
  - nan - serialport - bindings - rfxcom - node-red-contrib-rfxcom 
```

fivdi, 2020-07-24:
>I'm running on an Intel NUC with Ubuntu 20.04 LTS and Docker
>I'm not sure why Raspberry Pi is mentioned in the logs

Ok. It's probably not important.

I'm afraid I'm not familiar with configuring Node-RED and Docker images.

>prebuild-install WARN install No prebuilt binaries found (target=12.17.0 runtime=node arch=x64 libc=musl platform=linux)
>gyp ERR! build error
>gyp ERR! stack Error: not found: make

All I can see from the error messages is that the `make` utility is not available when an attempt is being made to build serialport from source code. You'll need to figure out how to ensure the prerequisites for building Node native modules are available. As mentioned above the prerequisites are (to the best of my knowledge) make, g++ and python.

> I also tried specifying "make" in the config file along with some other modules which have been recommended on the Home Assistant forum, but this hasn't worked either

What is the exact error message displayed after specifying make in the config file?

maxwellhadley, 2020-07-24:
Could it be a user permissions thing?

jag100UK, 2020-07-24:
A user on the Home Assistant community has given me a solution. 

The following had to be added to my node-red configuration

```
system_packages:
  - make
  - python3
  - g++
  - gcc
  - linux-headers
npm_packages:
  - nan - serialport - node-red-contrib-rfxcom
init_commands: []
```
NodeRed is showing errors in the log on startup, but its working.

These are the errors..

```

Executing busybox-1.31.1-r16.trigger,
: 140 MiB in 52 packages,
(1/1) Installing make (4.3-r0),
(1/12) Installing binutils (2.34-r1),
(2/12) Installing gmp (6.2.0-r0),
(4/12) Installing libgomp (9.3.0-r2),
(3/12) Installing isl (0.18-r0),
(5/12) Installing libatomic (9.3.0-r2),
(6/12) Installing libgphobos (9.3.0-r2),
(8/12) Installing mpc1 (1.1.0-r1),
(9/12) Installing gcc (9.3.0-r2),
(7/12) Installing mpfr4 (4.0.2-r4),
(10/12) Installing musl-dev (1.1.24-r9),
(12/12) Installing g++ (9.3.0-r2),
(11/12) Installing libc-dev (0.7.2-r3),
: 333 MiB in 64 packages,
(1/1) Installing linux-headers (5.4.5-r1),
: 340 MiB in 65 packages,
OK: 28525 distinct packages available,
[11:31:05] INFO: Starting installation of custom NPM/Node-RED packages...,
> @serialport/bindings@9.0.0 install /opt/node_modules/@serialport/bindings,
> prebuild-install --tag-prefix @serialport/bindings@ || node-gyp rebuild,
prebuild-install WARN install No prebuilt binaries found (target=12.17.0 runtime=node arch=x64 libc=musl platform=linux),
make: Entering directory '/opt/node_modules/@serialport/bindings/build',
  CXX(target) Release/obj.target/bindings/src/serialport.o,
In file included from ../src/./serialport.h:6,,
                 from ../src/serialport.cpp:1:,
../../../nan/nan.h: In function 'void Nan::AsyncQueueWorker(Nan::AsyncWorker*)':,
../../../nan/nan.h:2294:62: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type],
 2294 |     , reinterpret_cast<uv_after_work_cb>(AsyncExecuteComplete),
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Open(Nan::NAN_METHOD_ARGS_TYPE)':,
      |                                                              ^,
../src/serialport.cpp:78:69: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type],
   78 |   uv_queue_work(uv_default_loop(), req, EIO_Open, (uv_after_work_cb)EIO_AfterOpen);,
      |                                                                     ^~~~~~~~~~~~~,
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Update(Nan::NAN_METHOD_ARGS_TYPE)':,
../src/serialport.cpp:135:71: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type],
  135 |   uv_queue_work(uv_default_loop(), req, EIO_Update, (uv_after_work_cb)EIO_AfterUpdate);,
      |                                                                       ^~~~~~~~~~~~~~~,
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Close(Nan::NAN_METHOD_ARGS_TYPE)':,
../src/serialport.cpp:175:70: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type],
  175 |   uv_queue_work(uv_default_loop(), req, EIO_Close, (uv_after_work_cb)EIO_AfterClose);,
      |                                                                      ^~~~~~~~~~~~~~,
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Flush(Nan::NAN_METHOD_ARGS_TYPE)':,
../src/serialport.cpp:215:70: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type],
  215 |   uv_queue_work(uv_default_loop(), req, EIO_Flush, (uv_after_work_cb)EIO_AfterFlush);,
../src/serialport.cpp: In function 'Nan::NAN_METHOD_RETURN_TYPE Set(Nan::NAN_METHOD_ARGS_TYPE)':,
../src/serialport.cpp:270:68: warning: cast between incompatible function types from 'void (*)(uv_work_t*)' {aka 'void (*)(uv_work_s*)'} to 'uv_after_work_cb' {aka 'void (*)(uv_work_s*, int)'} [-Wcast-function-type],
  270 |   uv_queue_work(uv_default_loop(), req, EIO_Set, (uv_after_work_cb)EIO_AfterSet);,
[... 51 lines cut ...]
> serialport@9.0.0 postinstall /opt/node_modules/serialport,
Thank you for using serialport!,
If you rely on this package, please consider supporting our open collective:,
> https://opencollective.com/serialport/donate,
+ serialport@9.0.0,,
1,
+ node-red-contrib-rfxcom@2.11.1,
+ nan@2.14.1,
added 18 packages from 14 contributors and audited 1077 packages in 8.016s,
37 packages are looking for funding,
  run `npm fund` for details,
found 4 low severity vulnerabilities,
  run `npm audit fix` to fix them, or `npm audit` for details,
[cont-init.d] user.sh: exited 0.,
[cont-init.d] done.,
[services.d] starting services,
[services.d] done.,
[11:31:14] INFO: Starting Node-RED...,
> addon-node-red@ start /opt,
> node $NODE_OPTIONS node_modules/node-red/red.js "--settings" "/etc/node-red/config.js",
24 Jul 11:31:14 - [info] ,
===================,
Welcome to Node-RED
```
Thank you to all who have helped me

fivdi, 2020-07-24:
The messages that are now being shown are warnings rather than errors and can be ignored.

closed 2020-08-04 by fivdi