serialport/node-serialport #2112: DLL errror on NW.JS

JurijTSL, 2020-07-24:
### Summary of Problem

(Please answer all 3)

- What are you trying to do? Read serial data sent from Arduino in NW.JS app
- What happens?
It throws an erro and nothing happensr:
[image]

- What should have happened? Read serial data.

### Code to Reproduce the Issue

```js
    const SerialPort = require('serialport');
    const Readline = require('@serialport/parser-readline');
    const port = new SerialPort('COM3', { baudRate: 9600 });
    const parser = port.pipe(new Readline({ delimiter: '\n' }));
    // Read the port data
    port.on("open", () => {
        console.log('serial port open');
    });
    parser.on('data', data =>{
        console.log('got word from arduino:', data);
    });
```

### Versions, Operating System and Hardware

- SerialPort@? COM3
- Node.js v? v12.13.1
- Windows? Linux? Mac? Windows 10.
- Hardware and chipset? (Prolific/FTDI/Other) Arduino

closed 2020-10-04 by stale[bot]

Kalmon, 2021-08-31:
I have the same problem ....

LucianoPerezMontalvan, 2024-02-27:
Same problem here, i think is something bout prebuild, but can't find a solution