serialport/node-serialport #2114: Not available anymore in releases - "serialport-v4.0.7-node-v57-darwin-x64.tar.gz"

jameshone, 2020-07-29:
### Summary of Problem

- What are you trying to do?
I am having trouble installing your module into my build. The module for Darwin is not in the repo. 
See : [v4.0.7 - no Darwin available](https://github.com/serialport/node-serialport/releases/tag/4.0.7)

- What happens?
I get a 404 not found.

- What should have happened?
I should be able too dowload the tarball and install the module.

### Code to Reproduce the Issue

```bash
> node-pre-gyp install --fallback-to-build

node-pre-gyp info it worked if it ends with ok
node-pre-gyp verb cli [ '/Users/{user}/.nvm/versions/node/v8.6.0/bin/node',
node-pre-gyp verb cli   'app/node_modules/serialport/node_modules/.bin/node-pre-gyp',
node-pre-gyp verb cli   'install',
node-pre-gyp verb cli   '--fallback-to-build' ]
node-pre-gyp info using node-pre-gyp@0.6.32
node-pre-gyp info using node@8.6.0 | darwin | x64
node-pre-gyp verb command install []
node-pre-gyp info check checked for "app/node_modules/serialport/build/Release/serialport.node" (not found)
node-pre-gyp http GET https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz
node-pre-gyp http 404 https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz
node-pre-gyp ERR! Tried to download(404): https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz 
node-pre-gyp ERR! Pre-built binaries not found for serialport@4.0.7 and node@8.6.0 (node-v57 ABI) (falling back to source compile with node-gyp) 
node-pre-gyp http 404 status code downloading tarball https://github.com/EmergingTechnologyAdvisors/node-serialport/releases/download/4.0.7/serialport-v4.0.7-node-v57-darwin-x64.tar.gz 
```

### Versions, Operating System and Hardware

- SerialPort@4.0.7
- Node.js v8.6.0
- Mac

reconbot, 2020-08-01:
Sorry I can't support v4 any longer as I don't have any control over the EmergingTechnologyAdvisors github and it's several years out of date with some critical bugs.

reconbot, 2020-08-01:
If I recall correctly you might be able to download the binaries that you linked and put them in a folder that [bindings](https://www.npmjs.com/package/bindings) can find.

closed 2020-08-01 by reconbot