serialport/node-serialport #2115: Problem with baudRate

tobschoe, 2020-08-04:
### Summary of Problem

- What are you trying to do?
Setting a baudRate that is not 9600. For example 19200

- What happens?
The baudRate is set to the default 9600

- What should have happened?
The baudRate should be changed to my entered value

### Code to Reproduce the Issue
`const SerialPort = require('serialport')
const config = require('./config/config')

const port = new SerialPort(config.serialpath, function (err) {
  if (err) {
    return console.log('Error: ', err.message)
  }
},

{
  baudRate: config.baudrate,
  dataBits: 8,
  stopBits: 1,
  Parity: 'none',
  flowControl: false
}
)

  port.on('open', function () {
    console.log('port open. Data rate: ' + port.baudRate)
  })
`
### Versions, Operating System and Hardware
Node v12.16.3

node-serialport version (v9.0.0 )

Tried it on Windows 10 64 bit and Raspberry Pi model b+running Raspberry Pi OS

fivdi, 2020-08-04:
The options should be passed as the second argument to the constructor rather than as the third argument in the sample code above. In other words, this should work:
```js
const SerialPort = require('serialport')
const config = require('./config/config')

const port = new SerialPort(config.serialpath, {
  baudRate: 19200,
  dataBits: 8,
  stopBits: 1,
  Parity: 'none',
  flowControl: false
}, function (err) {
  if (err) {
    return console.log('Error: ', err.message)
  }
})

port.on('open', function () {
  console.log('port open. Data rate: ' + port.baudRate)
})
```

tobschoe, 2020-08-04:
It worked. Thanks a lot!

closed 2020-08-04 by tobschoe