serialport/node-serialport #2117: Serialport.on('data') stops reading data / stops working after some time

tobschoe, 2020-08-05:
### Summary of Problem

(Please answer all 3)

- What are you trying to do?
Im using Serialport to communicate with a radio transceiver. After some amount of time (happens on minute one, sometimes on minute 30 running this programm) the on data event doesnt contain any data even tho the event occures.

Restarting the transceiver doesnt fix the problem, i have to restart the node process for it to work again.

I tried using both port.on data and parser.on data.
( https://github.com/serialport/node-serialport/issues/2068  states that using port.on data resolves the problem for them, but not for me )

- What happens?
[image]

- What should have happened?
More data (like: fefe0094001094312700fd) should have been displayed.
### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')
const config = require('../config/config')

const port = new SerialPort(config.serialpath,
{
  baudRate: config.baudrate,
  dataBits: 8,
  stopBits: 1,
  Parity: 'none',
  flowControl: false
  },
  function (err) {
    if (err) {
      return console.log('Error: ', err.message)
    }
  },
  
)

port.on('data', function (data) {
    let dataString = Buffer.from(data).toString('hex')
    console.log(dataString)
})

// Code
```

### Versions, Operating System and Hardware

- SerialPort@? v.9
- Node.js v? v12.16.3
- Windows? Linux? Mac? Windows and Linux
- Hardware and chipset? (Prolific/FTDI/Other) Raspberry Pi b+

tobschoe, 2020-08-06:
update:
The problem seems to be that the on data event doesnt execute the callback function after a random amount time.
This doesnt solve the problem, but the next question is why wont the callback execute

reconbot, 2020-08-07:
What version of serialport are you using?

fivdi, 2020-08-07:
Which serial port on the Raspberry Pi B+ is being used?

Is `/dev/serial0` (or `/dev/ttyAMA0`) being used? This is the device corresponding to the serial port broken out to pins 8 and 10 on the Raspberry Pi header.

If `/dev/serial0` (or `/dev/ttyAMA0`) is being used, has the login shell which is accessible over this serial port been disabled? The default behaviour on Raspberry Pi OS is for a login shell to be accessible over this serial port.

If the login shell and the JavaScript program attempt to use this serial port at the same time it will not function correctly.

The login shell which is accessible over this serial port can be disabled from the command line with the `raspi-config` tool. 

Run the command `sudo raspi-config` and navigate to "5 Interfacing Options" > "P6 Serial".

Answer the question "Would you like a login shell to be accessible over serial?" with  No

Answer the question "Would you like the serial port hardware to be enabled?" with Yes.

Then reboot the Pi.

tobschoe, 2020-08-07:
@reconbot : I am using the latest version: 9.0.0

@fivdi : on windows `'COM3'` and on teh raspberry: `'/dev/ttyUSB0'`. The device is connected via USB

Even tho the device is connected i disabled the login shell via serial port in the `raspi-config`.

It seems to work when i call the `port.resume()` on every data event callback. ( need futher investigations. )
I dont know if this is a bad practise.

fivdi, 2020-08-07:
To me, the program from the first post above looks like it should work correctly.

Try the following: on the Raspberry Pi, run the following command to see if there are similar communication issues:
```
minicom -b 19200 -D /dev/ttyUSB0 -H --wrap
```
Change the baudrate if required. If `minicom` has not been installed already it can be installed with the following command:
```
sudo apt-get install minicom
```
If there are communication issues using minicom, it's more likely to be a device issue than a serialport issue.

Another idea would be to to add the following event handlers to the program to see if there are errors:
```js
port.on('error', function(err) {
  console.log(err);
})

port.on('close', function(err) {
  console.log(err);
});
```

tobschoe, 2020-08-11:
@fivdi 

I am using `port.on('error')` and `close`. Never have i gotten an error event emitted. 

`port.on('close')` functions as it should when i close the port.

The problem got "resolved" by calling `port.resume()` on every `port.on('data')` callback.

closed 2020-10-22 by stale[bot]

barryvdh, 2020-10-24:
It looks like I'm having a similar issue, the serial connection seems to be lost after a while. Nothing happens anymore. After rebooting the script, all the old messages are drained from the buffer at once.

josegpulido, 2021-03-09:
@tobschoe `port.resume()` works like a charm. But definitively seems like not a good solution.

josegpulido, 2021-03-09:
@barryvdh did you tried what @tobschoe said? or do you found another way?

LazyRichard, 2021-03-10:
I'm not sure this is the same issue or not, but I had a similar issue.
condition is that remove all parsers and if there is no parser on the port, then the serialport is stopping read.
my workaround is create dummy parser and insert into port and replace `port.on('data')` to `parser.on('data')`

```typescript
import stream from 'stream';

/**
 * If all readers are unpiped, the reader flow of port is stopped.
 * so, cannot retrieve data anymore.
 * To fix this, create dummy reader that just push all the chunk in the buffer.
 */
export class DefaultParser extends stream.Transform {
  constructor(options?: stream.TransformOptions) {
    super(options);
  }

  // eslint-disable-next-line
  _transform(chunk: any, encoding: any, cb: any): void {
    this.push(chunk, encoding);

    cb();
  }
}
```

joecorsi, 2021-11-05:
i can confirm port.resume() inside a port.on('data') call eliminates this issue. without it i see the serial connection dying within 30seconds to 1 minute.

ceremcem, 2022-07-30:
`port.resume()` in `parser.on('data')` didn't solve the problem in my case. I also have the very similar issue with Cutecom, which I'm using for years now. So my problem might not the same as the others.

elRadish, 2022-12-21:
Hi together, 
today, I faced the same issue and alle the above tips didn't solve it.
I was working with a Calliope mini, similar to BBC Microbit that has a Segger JLink USB Interface.
It is working with the Microsoft USB-Serial drivers, but shows the effect that the data-callback is stopping after a while.
After installing the Segger Jlink Driver, it works flawlessly.

yyberi, 2022-12-22:
I can confirm this same issue. `port.on('data')` callback is not called after ~5-60 minutes random time.
`port.resume()` didn't help. I tested the communication with minicom and it works perfectly several days in a row.
I'm having USB adapter in Raspberry and node-serialport version is 10.4.

yyberi, 2022-12-29:
I reinstalled serialport and it was updated to 10.5.0 and now it works like a charm!
With earlier 10.4 version it worked only if I had a Readline parser, but still unstable.