serialport/node-serialport #2118: Do you will support node-serialport on 32bit machines?

Ks89, 2020-08-06:
### Summary of Problem

I saw that with node-serialport 9 you dropped 32 bit support. This is perfectly understandable in case of desktops and servers, however many ARM embedded devices still have 32 bits CPUs.

If not, do you will support node-serialport 8.x.x on 32bit machines?

### Versions, Operating System and Hardware

- SerialPort@? 8.x.x
- Node.js v? 10
- Windows? Linux? Mac? Linux (distro for embedded devices)
- Hardware and chipset? ARM 32 bit embedded device

fivdi, 2020-08-06:
@Ks89 which ARM 32 bit embedded device are you using?

Ks89, 2020-08-06:
I'm using a board with NXP iMX6.

reconbot, 2020-08-06:
We've never provided builds or tested on ARM. I don't know why it wouldn't work but we don't have the infrastructure to prove it. Since nodejs dropped 32bit support we can't easily test against it either.

I don't intend on breaking 32bit support, but we won't know if it does break.

closed 2020-08-06 by Ks89

fivdi, 2020-08-06:
serialport 9 dropped Linux / ia32 (Intel Architecture, 32-bit) from its testing and build matrix. As shown in the [Supported Platforms and Architectures](https://serialport.io/docs/guide-platform-support#supported-platforms-and-architectures) section of the documentation, Windows / x86 (also 32-bit) is still supported.

serialport never officially supported any of the ARM architectures. The documentation simply states that they will probably work.

This means that serialport 9 will probably work on an NXP iMX6 (Armv7-A, 32-bit). Or are you experiencing issues here?

fivdi, 2020-08-06:
> Since nodejs dropped 32bit support ...

While this is true for Linux / ia32, nodejs still officially support Windows x86 and Linux armv7 which are also 32-bit.

reconbot, 2020-08-06:
I have to check appveyor but we might have had other issues with windows x86. If it's still in the matrix then we can support it, and should update the docs.

I put a lot of hours into a qemu arm docker build solution and haven't been successful. I'd pay for it if I could. I know the folks at belena have it working for them but it's not something we can just use out of the box.

reconbot, 2020-08-07:
Does look like we're still building win32 x86 however it _is_ failing randomly on appveyor.

Ks89, 2020-08-07:
> > Since nodejs dropped 32bit support ...
> 
> While this is true for Linux / ia32, nodejs still officially support Windows x86 and Linux armv7 which are also 32-bit.

Oh perfect. My Arm device is an ARMv7 32 bit.

Ks89, 2020-08-07:
I tried to install node-serialport 9.0.0 and it worked perfectly. 👍 
I opened this issue only to understand the future of the project.

dmcissokho, 2020-10-15:
> I tried to install node-serialport 9.0.0 and it worked perfectly. 👍
> I opened this issue only to understand the future of the project.

Did you installed it directly on the device or did you cross-compiled it? I also work on a Linux armv7  but with no internet connexion

Ks89, 2020-10-15:
> > I tried to install node-serialport 9.0.0 and it worked perfectly. 👍
> > I opened this issue only to understand the future of the project.
> 
> Did you installed it directly on the device or did you cross-compiled it? I also work on a Linux armv7 but with no internet connexion

I installed on the device.

dmcissokho, 2020-10-23:
O

> > > I tried to install node-serialport 9.0.0 and it worked perfectly. 👍
> > > I opened this issue only to understand the future of the project.
> > 
> > 
> > Did you installed it directly on the device or did you cross-compiled it? I also work on a Linux armv7 but with no internet connexion
> 
> I installed on the device.

Ok thank you, I tried but I have a g++  error, did you use any parameters on your npm install ?

Ks89, 2020-10-23:
I simply run npm install. For sure it's something missing on your embedded os (some dev dependency to build c stuff but I'm not an expert on this topic).
I can't send the node modules folder, it's a project of my company.

dmcissokho, 2020-10-23:
> I simply run npm install. For sure it's something missing on your embedded os (some dev dependency to build c stuff but I'm not an expert on this topic).
> I can't send the node modules folder, it's a project of my company.

Thank you 👍