serialport/node-serialport #2119: Cannot install node-serialport - ERR! 404 Not Found @serialport/binding-abstract

Ks89, 2020-08-06:
### Summary of Problem

When I try to install node-serialport I get this error:

npm ERR! code E404
npm ERR! 404 Not Found - GET https://npm.pkg.github.com/freedompro/@serialport%2fbinding-abstract - Package [@serialport/binding-abstract] was not found
npm ERR! 404
npm ERR! 404  '@serialport/binding-abstract@^8.0.6' is not in the npm registry.
npm ERR! 404 You should bug the author to publish it (or use the name yourself!)
npm ERR! 404 It was specified as a dependency of '@serialport/binding-mock'
npm ERR! 404
npm ERR! 404 Note that you can also install from a
npm ERR! 404 tarball, folder, http url, or git url.

### Versions, Operating System and Hardware

- SerialPort@? 8.0.6
- Node.js v? 10
- Windows? Linux? Mac? Linux 32 bit for embedded devices
- Hardware and chipset? (Prolific/FTDI/Other) ARM v7 32 bit Cpu

RUGMJ7443, 2020-08-06:
I'm having the same issue but with installing binding-mock

Ks89, 2020-08-06:
It was probably caused by with npmjs. Status page reported a service incident some hours ago.
I close the issue, because it seems not caused by node-serialport

closed 2020-08-06 by Ks89