serialport/node-serialport #2125: SerialPort v9.0.1 with Electron - Windows ia32

juliocesaroliveira, 2020-08-10:
Not able to build Electron app (Windows ia32) with SerialPort v9.0.1

I am using serialport v9.0.1 in my Electron app and building Linux(x64) and Windows(ia32 and x64) via electron-builder.

"electron": "^9.1.2",
"electron-builder": "^22.8.0",
"serialport": "^9.0.1"

When I try to build electron app for Windows ia32, it throws this error:
```
http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.1/bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
```

```
• rebuilding native dependencies  dependencies=@serialport/bindings@9.0.1 platform=win32 arch=ia32
• install prebuilt binary  name=@serialport/bindings version=9.0.1 platform=win32 arch=ia32
  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 5.3.5
    prebuild-install info looking for cached prebuild @ /home/julio/.npm/_prebuilds/cca835-bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.1/bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.1/bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.1.2 runtime=electron arch=ia32 libc= platform=win32)
    
                                    command=/home/julio/.nvm/versions/node/v12.10.0/bin/node /home/julio/development/consistem/js-projects/csw_browser/app/node_modules/prebuild-install/bin.js --platform=win32 --arch=ia32 --target=9.1.2 --runtime=electron --verbose --force
                                    workingDir=/home/julio/development/consistem/js-projects/csw_browser/app/node_modules/serialport/node_modules/@serialport/bindings
error Command failed with exit code 1.
```

### Versions, Operating System and Hardware

- SerialPort@9.1.2
- Node.js v12.10.0
- Building for Windows ia32 using Linux with Wine(Ubuntu 5.5-3ubuntu1)

reconbot, 2020-08-11:
I wonder if there's a log file output or something that gives more light to

>  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible

juliocesaroliveira, 2020-08-11:
npm console log:
```
npm run --verbose dist:win
npm info it worked if it ends with ok
npm verb cli [
npm verb cli   '/home/julio/.nvm/versions/node/v12.10.0/bin/node',
npm verb cli   '/home/julio/.nvm/versions/node/v12.10.0/bin/npm',
npm verb cli   'run',
npm verb cli   '--verbose',
npm verb cli   'dist:win'
npm verb cli ]
npm info using npm@6.10.3
npm info using node@v12.10.0
npm verb run-script [ 'predist:win', 'dist:win', 'postdist:win' ]
npm info lifecycle Consistem@7.0.1-beta.75~predist:win: Consistem@7.0.1-beta.75
npm info lifecycle Consistem@7.0.1-beta.75~dist:win: Consistem@7.0.1-beta.75

> Consistem@7.0.1-beta.75 dist:win /home/julio/development/consistem/js-projects/csw_browser/app
> electron-builder --win

  • electron-builder  version=22.8.0 os=5.4.0-42-generic
  • loaded configuration  file=package.json ("build" field)
  • loaded parent configuration  file=/home/julio/development/consistem/js-projects/csw_browser/app/node_modules/electron-webpack/out/electron-builder.js
  • electron-rebuild not required if you use electron-builder, please consider to remove excess dependency from devDependencies

To ensure your native dependencies are always matched electron version, simply add script `"postinstall": "electron-builder install-app-deps" to your `package.json`
  • writing effective config  file=dist/builder-effective-config.yaml
  • rebuilding native dependencies  dependencies=@serialport/bindings@9.0.1 platform=win32 arch=ia32
  • install prebuilt binary  name=@serialport/bindings version=9.0.1 platform=win32 arch=ia32
  ⨯ cannot build native dependency  reason=prebuild-install failed with error and build from sources not possible because platform or arch not compatible
                                    cause=exit status 1
                                    errorOut=prebuild-install info begin Prebuild-install version 5.3.5
    prebuild-install WARN install prebuilt binaries enforced with --force!
    prebuild-install WARN install prebuilt binaries may be out of date!
    prebuild-install info looking for cached prebuild @ /home/julio/.npm/_prebuilds/cca835-bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
    prebuild-install http request GET https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.1/bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
    prebuild-install http 404 https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.1/bindings-v9.0.1-electron-v80-win32-ia32.tar.gz
    prebuild-install WARN install No prebuilt binaries found (target=9.1.2 runtime=electron arch=ia32 libc= platform=win32)
    
                                    command=/home/julio/.nvm/versions/node/v12.10.0/bin/node /home/julio/development/consistem/js-projects/csw_browser/app/node_modules/prebuild-install/bin.js --platform=win32 --arch=ia32 --target=9.1.2 --runtime=electron --verbose --force
                                    workingDir=/home/julio/development/consistem/js-projects/csw_browser/app/node_modules/serialport/node_modules/@serialport/bindings
npm verb lifecycle Consistem@7.0.1-beta.75~dist:win: unsafe-perm in lifecycle true
npm verb lifecycle Consistem@7.0.1-beta.75~dist:win: PATH: /home/julio/.nvm/versions/node/v12.10.0/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/home/julio/development/consistem/js-projects/csw_browser/app/node_modules/.bin:/home/julio/.nvm/versions/node/v12.10.0/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
npm verb lifecycle Consistem@7.0.1-beta.75~dist:win: CWD: /home/julio/development/consistem/js-projects/csw_browser/app
npm info lifecycle Consistem@7.0.1-beta.75~dist:win: Failed to exec dist:win script
npm verb stack Error: Consistem@7.0.1-beta.75 dist:win: `electron-builder --win`
npm verb stack Exit status 1
npm verb stack     at EventEmitter.<anonymous> (/home/julio/.nvm/versions/node/v12.10.0/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:326:16)
npm verb stack     at EventEmitter.emit (events.js:209:13)
npm verb stack     at ChildProcess.<anonymous> (/home/julio/.nvm/versions/node/v12.10.0/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
npm verb stack     at ChildProcess.emit (events.js:209:13)
npm verb stack     at maybeClose (internal/child_process.js:1021:16)
npm verb stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:283:5)
npm verb pkgid Consistem@7.0.1-beta.75
npm verb cwd /home/julio/development/consistem/js-projects/csw_browser/app
npm verb Linux 5.4.0-42-generic
npm verb argv "/home/julio/.nvm/versions/node/v12.10.0/bin/node" "/home/julio/.nvm/versions/node/v12.10.0/bin/npm" "run" "--verbose" "dist:win"
npm verb node v12.10.0
npm verb npm  v6.10.3
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! Consistem@7.0.1-beta.75 dist:win: `electron-builder --win`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the Consistem@7.0.1-beta.75 dist:win script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm verb exit [ 1, true ]
npm timing npm Completed in 1096ms
```
@serialport/bindings@9.0.1 doesn't have for the ia32 architecture, getting the exception http 404 (https://github.com/serialport/node-serialport/releases/download/@serialport/bindings@9.0.1/bindings-v9.0.1-electron-v80-win32-ia32.tar.gz Not Found)

When I build with electron@8.4.1 and serialport@9.0.0 it works fine, because there are bindings for ia32 architecture.

I think the error is due to not having bindings for the ia32 architecture.

reconbot, 2020-08-12:
it's becoming increasingly difficult to build win32 binaries as appveyor (our ci) seems to be having trouble with it. Building it yourself is probably the way to go.

juliocesaroliveira, 2020-08-13:
It was a little bit of work to create the environment to do the building.
I used node@13.8.0 to generate the binaries, because it was on Windows 7 and node @ 14 does not support it.
The @serialport resources I use, worked perfectly with electron@9.2.0.
Thank you for your support.

closed 2020-08-13 by juliocesaroliveira