serialport/node-serialport #2126: Cannot enumerate (list) serial ports in WSL

jeffrson, 2020-08-10:
### Summary of Problem

(Please answer all 3)

- What are you trying to do? execute SerialPort.list() in Debian WSL
- What happens? Promise resolves with empty array
- What should have happened? Array should include information about installed serial devices

### Code to Reproduce the Issue

```js
const SerialPort = require('serialport')

SerialPort.list().then(
    (ports) => console.log(ports),
    (err) => console.error(err)
)
```

However, when I just open /dev/ttyS3 (which corresponds to COM3) it works fine - data is sent and received.

### Versions, Operating System and Hardware

- SerialPort@? 9.0.1
- Node.js v? 14.7.0
- Windows? Linux? Mac? Windows - WSL (1)
- Hardware and chipset? (Prolific/FTDI/Other) FTDI

reconbot, 2020-08-11:
Can you run this with debug mode? (no idea if you need to use linux or windows debugging) https://serialport.io/docs/guide-debugging

jeffrson, 2020-08-11:
Unfortunately it seems this is not really helpful:
```
DEBUG=* node enum
  serialport/bindings loading LinuxBinding +0ms
  serialport/stream .list +0ms
[]
```

reconbot, 2020-08-11:
You nailed it. However you can require the windows binding directly and see if it's `list` method works.

If it does we may need to detect WSL somehow.

closed 2020-10-17 by stale[bot]

liangshen-hds, 2022-06-30:
I used usbipd to attach usb devices from Windows to WSL2. 
With "await SerialPort.list()" I got 

```
  {
    path: '/dev/ttyACM0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    locationId: undefined,
    vendorId: undefined,
    productId: undefined
  }
```

with "udevadm info -e"
```
P: /devices/platform/vhci_hcd.0/usb1/1-1
N: bus/usb/001/007
L: 0
E: DEVPATH=/devices/platform/vhci_hcd.0/usb1/1-1
E: SUBSYSTEM=usb
E: DEVNAME=/dev/bus/usb/001/007
E: DEVTYPE=usb_device
E: DRIVER=usb
E: PRODUCT=1546/1a6/703
E: TYPE=2/0/0
E: BUSNUM=001
E: DEVNUM=007
E: MAJOR=189
E: MINOR=6

P: /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0
L: 0
E: DEVPATH=/devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0
E: SUBSYSTEM=usb
E: DEVTYPE=usb_interface
E: DRIVER=cdc_acm
E: PRODUCT=1546/1a6/703
E: TYPE=2/0/0
E: INTERFACE=2/2/1
E: MODALIAS=usb:v1546p01A6d0703dc02dsc00dp00ic02isc02ip01in00

P: /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0/tty/ttyACM0
N: ttyACM0
L: 0
E: DEVPATH=/devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0/tty/ttyACM0
E: SUBSYSTEM=tty
E: DEVNAME=/dev/ttyACM0
E: MAJOR=166
E: MINOR=0

P: /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.1
L: 0
E: DEVPATH=/devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.1
E: SUBSYSTEM=usb
E: DEVTYPE=usb_interface
E: DRIVER=cdc_acm
E: PRODUCT=1546/1a6/703
E: TYPE=2/0/0
E: INTERFACE=10/0/255
E: MODALIAS=usb:v1546p01A6d0703dc02dsc00dp00ic0Aisc00ipFFin01
```

I tried https://github.com/MadLittleMods/node-usb-detection and it can get
```
  {
    locationId: 1,
    vendorId: 5446,
    productId: 422,
    deviceName: 'u-blox 6  -  GPS Receiver',
    manufacturer: 'u-blox AG - www.u-blox.com',
    serialNumber: '',
    deviceAddress: 7
  },
```